COVESA/dlt-viewer #16: New release?

jjardon, 2018-06-12:
There has been a lot of improvements in the last 2 years since v2.18; is there any plans to a new release?

Some people is asking for an update in the Arch package :) : https://aur.archlinux.org/packages/dlt-viewer/

jeremiah, 2018-06-12:
From my understanding, there is no planned new release in the pipeline. But that doesn't mean that one can be done, I'll ask the maintainers to review your request.

jjardon, 2018-06-13:
Hey @jeremiah !

Not expectations really, but I'd like to use a tag instead a random commit to update the Arch package linked above

AWWenzel, 2018-06-13:
Hi @jjardon, @jeremiah ,

i will talk to Gernot, who is doing the maintenance.
He already planned to create a 2.19.0 version.

    Alex

shadamik, 2018-08-10:
What about a deb package? Are there any recipes available?

jeremiah, 2018-08-10:
There are deb packages floating around. Because GENIVI builds its demonstrators from a source code build too we don't officially maintain deb packages but there are some built and we can provide some packaging artifacts. I'll try and gather that material.

jeremiah, 2018-08-10:
Here is the Ubuntu PPA: https://launchpad.net/~genivi-ubuntu/+archive/ubuntu/utopic.ppa 
Unfortunately that only holds dlt-daemon and I think most folks want a package of dlt-viewer. I don't think that should be that hard to create, I'll look into it.

aigarius, 2018-08-14:
I am planning on getting this in Debian as well. Not sure how hard the specifics of dlt-viewer and Qt will be, but help is for sure welcome.

jeremiah, 2018-08-21:
I'm traveling at the moment, but when I'm back I plan to help as much as possible with testing (Arch packages, Debian packages, building from source) and will try and be a bridge between upstream and the distros. Fortunately we have a DLT maintainer in this thread so I'm hopeful that we can get some good, working packages out of this. :-)

jjardon, 2018-08-22:
Thanks @jeremiah , a new tag would be enough in my case (Arch) :)

GeniviDLTmaintainer, 2019-06-26:
The current plan is to prepare a new official release 2.19.0 until end of july 2019.

alexmucde, 2019-07-25:
Released v2.19.0 now

closed 2019-07-25 by alexmucde