COVESA/dlt-viewer #24: Fail to bind with ALEV3 and ALEV4 - Linux and Open-AVB

davmann, 2019-03-17:
Hi, 
**my System is:**
- Debian 9.8 and Ubuntu 18.04.2 (fresh install, updated and fully working)
- Open-AVB (fresh install, updated and fully working)

**Version 2.19** 
- fail to connect and bind to Booster ALEV3
- fail to connect and bind to Booster ALEV4

**Version 2.18** 
- connect and bind to Booster ALEV3 but return no internal components logs
- connect and bind to Booster ALEV4 but return empty logs (only ECU and no BOOST)

**Version 2.17**
- fail on both Boosters

Any ideas on hot to solve?
Thanks!

GeniviDLTmaintainer, 2019-05-02:
I guess DLT protocol is not related to ALEV !?

davmann, 2019-05-02:
You're right. DLT seems nor yet implemented in 2021 boosters, but in 2018 yes.
Anyway, even in 2018, I still cannot see DLT logs.

Maybe something wrong in the installation process? Is there something I'm missing? 
These are my steps using a linux machine to compile the latest stable DLT Viewer:
* sudo apt install qt5-default
* mkdir ~/GENIVI
* cd ~/GENIVI
* git clone https://github.com/GENIVI/dlt-viewer.git
* cd dlt-viewer/
* git checkout c52fcd3
* mkdir build
* cd build
* qmake-qt5 ../BuildDltViewer.pro
* make
* sudo make install
* sudo ldconfig

Thanks!

GeniviDLTmaintainer, 2019-06-26:
Is this topic related to the DLT Viewer or could there be an issue with the target?
Is this an UDP connection issue?

AWWenzel, 2019-07-08:
Looks like a target specific issue. Please reopen if you find an issue in the DLT Viewer.

closed 2019-07-08 by AWWenzel