COVESA/dlt-viewer #27: DltMessage to raw dlt log for dlt-viewer

punkproger, 2019-04-18:
Hello.

Could you please explain how to convert DltMessage structure to raw dlt log?
I want to write to file DltMessages after I get them using dltClient and filter them by my rules,
to see these logs using dlt viewer?

 
```
typedef struct sDltMessage
{
    /* flags */
    int8_t found_serialheader;

    /* offsets */
    int32_t resync_offset;

    /* size parameters */
    int32_t headersize;    /**< size of complete header including storage header */
    int32_t datasize;      /**< size of complete payload */

    /* buffer for current loaded message */
    uint8_t headerbuffer[sizeof(DltStorageHeader)+
        sizeof(DltStandardHeader)+sizeof(DltStandardHeaderExtra)+sizeof(DltExtendedHeader)]; /**< buffer for loading complete header */
    uint8_t *databuffer;         /**< buffer for loading payload */
    int32_t databuffersize;

    /* header values of current loaded message */
    DltStorageHeader       *storageheader;  /**< pointer to storage header of current loaded header */
    DltStandardHeader      *standardheader; /**< pointer to standard header of current loaded header */
    DltStandardHeaderExtra headerextra;     /**< extra parameters of current loaded header */
    DltExtendedHeader      *extendedheader; /**< pointer to extended of current loaded header */
} DltMessage;
```

gunnarx, 2019-04-18:
Please explain, what do you mean by "raw dlt log"?   Will you decide yourself the format (and content) of that log?  What does that format look like?

punkproger, 2019-04-18:
@gunnarx 
> Please explain, what do you mean by "raw dlt log"? 
"raw dlt log" -- how DltMessage structure should be written in text file to open this file in dlt-viewer. 

> What does that format look like?
Any as I can imagine:) Different AppId, Content Id, etc.

> What type of filtering should be possible?
I think standard C/C++ approaches.

gunnarx, 2019-04-18:
>  "raw dlt log" -- how DltMessage structure should be written in text file to open this file in dlt-viewer.

OK, that's understandable.  

The export function in DLT-Viewer creates such a file doesn't it?  Maybe work backwards from there?
[dltexporter.cpp](https://github.com/GENIVI/dlt-viewer/blob/master/src/dltexporter.cpp)

This "raw" format seems to correspond to "FormatDlt" if I'm not guessing wrong:
```
typedef enum { FormatDlt,FormatAscii,FormatCsv,FormatClipboard,FormatClipboardPayloadOnly,FormatDltDecoded,FormatUTF8} DltExportFormat;
```
Maybe the maintainer (not me) has some additional tips.

punkproger, 2019-04-18:
@gunnarx can I using std::ofstream and dltMessage structure write to file(dlt format) log to see this one in dlt-viewer? Or I should only use Qt?

AWWenzel, 2019-04-23:
The DLT file format is specified in the AUTOSAR standard Diagnostic Log and Trace, which you can found on  the AUTOSAR webpage.
Example code how the DLT Log files are written can be found in the qdlt source code as part of the DLT Viewer and in DLT daemon projects on github.

gunnarx, 2019-04-23:
>  Example code how the DLT Log files are written can be found in the qdlt source code as part of the DLT Viewer and in DLT daemon projects on github.

I only found it in dlt-viewer (as I linked to above), but if you want to use non-Qt code, maybe it's easier to find the corresponding in dlt-daemon (Of course the answer to your question, @punkproger, is that it's *possible* to do with standard C++ code but you have to try it and see and work out the details).

Are you, by the way, looking to make an open-source project here, your ideas around filtering I mean?   @punkproger

punkproger, 2019-04-23:
The solution is to write header and data buffer of dltMessage.
Thanks.

closed 2019-04-23 by punkproger