COVESA/dlt-viewer #34: Unsigned integer not being decoded

vadrianc, 2019-06-12:
Observed in DLT Viewer 2.18.0

I got a DLT file containing 4 DLT packets, each containing in the payload a single unsigned integer argument, 8-bit, 16-bit, 32-bit and 64-bit long. DLT Viewer decodes only the 8-bit long unsigned integer. If the packets are saved individually to a file, DLT Viewer displays each one, when opening the corresponding file. 

See [unsigned_int_arg.zip](https://github.com/GENIVI/dlt-viewer/files/3280888/unsigned_int_arg.zip)

AWWenzel, 2019-06-25:
@vadrianc I tried your file with a Windows version of DLT Viewer 64 Bit. I can see all four lines. The first one has the value 18446744073709551615, the second 4294967295, the third 65535 and the last 255.
Perhaps you have some issues with the built in index cache. Please empty or disable the cache in the Settings.

vadrianc, 2019-07-08:
@AWWenzel I tried to reproduce it again without clearing the cache as you suggest, but I was able to see all 4 DLT lines. Since I'm unable to reproduce it again, I believe the ticket can be closed.

AWWenzel, 2019-07-08:
Cannot be reproduced.

closed 2019-07-08 by AWWenzel