COVESA/dlt-viewer #35: String invalid offset position calculation when binary data is used in preceding PDUs

RhavoX, 2019-06-18:
The string position in the message is calculated in a wrong way. 
To avoid this this line needs to be removed: (588) `offset += sizeof(unsigned short);` from nonverboseplugin. 
Now the string position is correctly calculated. 
Can you apply this simple fix or do you require a pull request?

AWWenzel, 2019-06-25:
@RhavoX I do not think this is an error. In the DLT standard strings are not null terminated strings. They consist of a 2 byte length information and then the string itself without null termination.
Why do you think this wrong? Perhaps your non verbose DLT file is not correct?

RhavoX, 2019-06-25:
I have verified it and it seems that you are right, thank you for your feedback.
Issue closed :)

closed 2019-06-25 by RhavoX