COVESA/dlt-viewer #36: "Could not connect to any X display." error when running dlt-viewer on a VPS

trevorlazarus, 2019-06-19:
After following the instructions to build dlt-viewer on an Ubuntu  EC2, I try to run dlt_viewer but am getting the following error:

`qt.qpa.screen: QXcbConnection: Could not connect to display
Could not connect to any X display.
`
I understand that this is because QT is expecting a display. Is there any work around for this. Has any one built a different version of dlt-viewer that can be run a VPS?

I'm trying to convert DLT files on a server and tried using dlt-convert, but this only outputs ASCII and I'm looking to use dlt-viewer's output to CSV feature.

gunnarx, 2019-06-19:
@trevorlazarus Not 100% clear what you want to achieve because it is a graphical program.  So what is the behavior you expect if you manage to run it on the VPS.  Does your VPS have a GUI (X-based or otherwise) or not?

If you want just a text-based command-line tool:  When building dlt-daemon there are some command-line tools created, such as **dlt-receive**.  Maybe that can be useful as a starting point.  For reference, this is the [source code(https://github.com/GENIVI/dlt-daemon/tree/master/src/console) I'm talking about.

gunnarx, 2019-06-19:
OK, I saw this now.  

> I'm trying to convert DLT files on a server and tried using dlt-convert, but this only outputs ASCII and I'm looking to use dlt-viewer's output to CSV feature.

In that case you still need to trigger it somehow I suppose.  So... well maybe you need to take some of the source code from either project and modify it to create what you want.  

There are some ways to trick a "headless" system into running X programs but I'm not sure if this gets you the automated solution you probably want.  Typically, installing a VNC server will allow you to run X programs without a real display.

AWWenzel, 2019-06-25:
@trevorlazarus you can run the DLT Viewer by command line, but the DLt viewer is still a GUI tool in the background. I had some plans long time ago to provide a version of DLT Viewer without GUI for some use cases like you have, but i hade no time so far to develope it. As Gunnar mentioned you have to write a new tool which uses the source code from DLT Viewer like qdlt library. Sorry that i have no solution for you at the moment.

GeniviDLTmaintainer, 2019-06-27:
I did install xvfb-run on ubuntu 16.04. So I can run e.g. commandline conversion in an automated setup. But there must be a graphical session of the user running on the machine ( it does not have to be active ).
In my case e.g.:
export DISPLAY=:1.0
xvfb.run dlt_viewer -s -c /testtraces/longtrace.dlt longtrace.txt

alexmucde, 2019-10-14:
We are trying to create a command line only version of DLT Viewer, but there is much effort to do so and we need some support and a lot of time here.
I will close this issue, because i think there is a short time solution from Gernot to fix this issue.

closed 2019-10-14 by alexmucde

yidaimi, 2022-04-13:
I encounted this problem in version 2.21.2 too. I want to use 'dlt-view -c' tool to convert file from dlt to file.
Unfortunatelly, the tool will be used in CentOS 7.6 server without display device. 
I found this and it works:
https://serverdiary.com/linux/how-to-install-xvfb-and-create-systemd-xvfb-service-on-centos-7-8-or-rhel-7-8/
By the way ,if it doesn't work, you should execute 'export DISPLAY=localhost:0.0' at the end.