COVESA/dlt-viewer #41: [DLT Plugin] How to decode message QDltFile?

svlad-90, 2019-07-01:
Hi everyone.

I'm writing my own plugin in order to analyse frequency of different kinds of messages from different applications ( trace spam analyser ).

In short, when client presses the "Analyse" button, my plugin needs to iterate over all messages within a dlt file, and pass them through regex in order to do a grouping of all processed messages.

For that purpose I can't use initMsgDecoded & updateMsgDecoded, as those are called only when messages are being added or updated. When those callbacks are called, the messages, which are passed there, are destroyed right afterwards, so I can't store the pointers. I need to do copies of all decoded messages. And in my case, that leads to dlt-viewer consuming about 5 Gb of RAM ( some huge files to be analysed ).

What I want to do, is to be able to iterate over the messages using QDltFile, which is passed in "initFileStart", cause QDltFile works with some cached file ( as far as I've got ). Thus, huge memory consumption will be avoided.

The issue here is that QDltFile has all messages encoded. And to be able to decode them I need to have access to QDLTPluginManager::decodeMsg, which can iterate over all plugins in order to have totally decoded message.

That QDLTPluginManager is part of mainwindow, and there is no way to access it inside the plugin without some ugliest workarounds. E.g. iterating over root widgets of qApp in order to find main window instance. The dltpluginmanager is inside public section of mainwindow, thus it is possible to do that. Still, that is too ugly.

May be I've missed something?
Is there other way to go?

May be DLTPluginManager be wrapped into a singleton facade, which will have a "decodeMsg" method in its public section? That will not allow plugin implementation to access critical parts of plugin manager, still deconding of messages will still be possible. What do you think?

Somehow other parts of dlt viewer core has such access, e.g. searchdialog. But plugin does not have it.

svlad-90, 2019-07-01:
I was able to find a way out. 
Still it is a work-around.
But, it works.

I've created a second instance of QDLTPluginManager within my own plugin, and managed to load plugins once again.
 
As my call to load the plugins is the second one ( comes after the core's one ), my call does not actually  load *.dll plugins again, as they are already loaded. 

Moreover, as QDLTPluginManager hides all instances of plugins under QDltPlugin class, I'm actually sharing the same instances of QDLTPluginInterface ( and it relatives ) with the core's QDLTPluginManager.

That allows me to easily get instance to required plugin.
E.g. by it's type, like "decoder plugins", or by name, like "specific decoder plugin".
And that is without creating duplicates of plugins, forms, etc.

As plugin runs in the same thread, as the core, there are no race conditions.

Moreover, core is still responsible for destruction of all the plugins. At least I assume so, even if the only place, where I've found "unload" in this repository will be never executed. Most probably loaded plugins are destroyed together with application.

In any case, there is no chance, that I will try to call some decoder plugin inside my own plugin's implementation, when that plugin is dead, as most probably, I will be also already destroyed.

The only thing I need to fulfil - use neighbour plugins ONLY from the same thread, and avoid usage of the neighbour plugins in my plugin's destructor.

As I'm going to pass long term regex processing to worker-threads ( in order to unblock UI ), I'll still need to prepare all the data in the main thread, and only then pass prepared portions of messages to worker-thread's context.

Not the best solution, but reliably it works.
If there exist some more proper official solution - let me know and I'll switch to it ☺

gunnarx, 2019-07-01:
This is an interesting use case first of all, and I think others might be interested too.

I therefore feel like these kinds of exploratory design discussions would be better handled on a mailing list.  GENIVI has a dedicated list for this project:  [genivi-diagnostic-log-and-trace](https://lists.genivi.org/mailman/listinfo/genivi-diagnostic-log-and-trace_lists.genivi.org) and also the general [genivi-projects list](https://lists.genivi.org/mailman/listinfo/genivi-projects_lists.genivi.org).   I don't know how much the maintainers are driving discussions to those lists (Please consider doing that) but the DLT list is listed in the README (but no specific instructions when/how to use it).  

If discussions are driven to those lists, it's of course important to have maintainers answering to emails unless there is an answer from other community members.

Just my thoughts on this.

svlad-90, 2019-07-01:
Hi @gunnarx, thanks for your reply.

Actually I have my use-case solved.

But I'll do the following:
- I'll wait for several days to get some answers from maintainers here.
- If that will not work, I will close this issue and send message to the suggested mailing list ( genivi-diagnostic-log-and-trace ).

Hopefully one of 2 above points will allow to extend the plugin's API by allowing direct access to message decoding capabilities ☺

gunnarx, 2019-07-01:
Note, be aware of mailing list link change - see #42

AWWenzel, 2019-07-08:
Hi @svlad-90,

you are right. There is currently no possibility to access decoded messages via the plugin interface.
You can only access the DLT messages of the current open DLT file by the QDltFile object you will get with the Plugin Callback virtual void initFileStart(QDltFile *file).

Some years ago before i stopped working on DLT Viewer i started to restructure the Plugin and decode message handling and move everything into qdlt library, but i did not managed to finish this work.

I think your quick solution sounds good and we can have a look at it and perhaps integrate it, if you provide a pull request.

Best regards

     Alex

svlad-90, 2019-07-15:
Hi @ghost,

I've just seen your message.
My quick solution is still a workaround. 
It can be used by a dedicated plugin, but doesn't worth to be included to the core.

The thing which might be useful is addition of the new injection method to the plugin's interface:

virtual void injectDecoderPlugins( const QList<QDLTPluginDecoderInterface>& decoderPlugins ) = 0;

or it can be even a wrapper, to protect the plugin from accessing the other plugin's pointers:

virtual void injectDecoderFacade( const QDLTPluginDecoderFacade& decoderFacade ) = 0;

Still that would mean an increment of the PLUGIN_INTERFACE_VERSION, which might mean, that all previously existing plugins will need to be rebuild. If that is acceptable, I can try to prepare the change.

alexmucde, 2019-07-15:
A rebuild of the plugins is no problem in a new release of DLT Viewer.
Yes please provide a patch. We can add it to the next version v2.20.0

svlad-90, 2019-08-11:
Hi everyone, 
I've prepared a change, but I can't push it to this repository ( even to a separate branch ).
It seems, that I need to be added to a list of contributors in order to propose my PR.
Can someone grant me required rights?

alexmucde, 2019-08-12:
Hi @svlad-90!

Only maintainers have write access to the GENIVI repository.
You have to create a personal clone of DLT Viewer first, then push your changes to a branch there and then create the pull request.

svlad-90, 2019-08-12:
Hi @alexmucde, Ok. Got it.
PR was created - https://github.com/GENIVI/dlt-viewer/pull/49/commits
Please, have a look.

alexmucde, 2019-10-18:
PR merged.

closed 2019-10-18 by alexmucde