COVESA/dlt-viewer #45: [Building error] Unknown module(s) in QT: serialport

MingL0L, 2019-07-18:
Hello,
I know noting about QT, but i need run dlt viewer in linux.
So, I was tring to build dlt viewer from source code in ubuntu 18.04 following INSTALL.txt.
First, i installed QT using following command:
    sudo apt install build-essential
    sudo apt install qtcreator
    sudo apt install qt5-default

Then,   
    * mkdir build
    * cd build
    * qmake-qt5 ../BuildDltViewer.pro
    * make
Then i got this building error:
    **Unknown module(s) in QT: serialport** 

Can i get some help, please?

[err_log.txt](https://github.com/GENIVI/dlt-viewer/files/3407745/err_log.txt)

MingL0L, 2019-07-18:
Actually, i just need to convert .dlt file to text file. Is there some easier way?

alexmucde, 2019-07-19:
Hello @MingL0L ,

you need the package libqt5serialport5-dev in Ubuntu 18.04 to compile and install the DLT Viewer.

sudo apt install libqt5serialport5-dev

I hope this helps.

    Alex

alexmucde, 2019-07-19:
I will also update the INSTALL.txt.
Thanks for your Feedback.

   Alex

alexmucde, 2019-07-25:
Fixed in INSTALL.txt. Please reopen, if still not working.

closed 2019-07-25 by alexmucde