COVESA/dlt-viewer #46: Unknown CMake command "QT5_WRAP_UI"

Mehmetcanss, 2019-07-19:
I am not a cmake user, but I did the following in MacOs, which gave me the following error:

brew install qt
Qt5_DIR="/usr/local/opt/qt" cmake src/

Gives the following error:

CMake Error at CMakeLists.txt:22 (QT5_WRAP_UI):
  Unknown CMake command "QT5_WRAP_UI".

syl22-00, 2019-07-19:
You are not pointing to the top level CMakeLists.txt file. you should point to the root folder of the source code, not the src/ folder.

Mehmetcanss, 2019-07-19:
Thanks, it worked but only after I exported the QtDIR in my bash_profile

closed 2019-07-19 by Mehmetcanss

sagaidakovskyi, 2019-12-04:
> You are not pointing to the top level CMakeLists.txt file. you should point to the root folder of the source code, not the src/ folder.

I have exported qt dir to bash profile:
export PATH="$PATH:/usr/local/opt/qt/bin"

However it is not clear what do you mean by "root folder of the source code". Is it directory above src/ ?

I am still getting Unknown CMake command "QT5_WRAP_UI"

sagaidakovskyi, 2019-12-05:
It appeared that I had to unzip dlt-viewer from scratch, because cmake cache prevented from correct build of application on macOS

However here are my parameters from bash_profile in case anyone might require them:

export CMAKE_PREFIX_PATH=/usr/local/Cellar/qt/5.13.2/
export Qt5_DIR="/usr/local/Cellar/qt/5.13.2/"
export QtDIR="/usr/local/Cellar/qt/5.13.2/"

sarndt, 2020-12-02:
> It appeared that I had to unzip dlt-viewer from scratch, because cmake cache prevented from correct build of application on macOS
>
> However here are my parameters from bash_profile in case anyone [...]

Thanks for this, this actually helped me, I ran into the same issue.
It wasn't necessary for me to clone the project anew or anything, but I needed to delete the `CMakeCache.txt` that was probably there created during one of my previous attempts to compile.