COVESA/dlt-viewer #50: [DLTViewer] Race condition in "initMsgDecoded" & "initMsg" callbacks

svlad-90, 2019-08-12:
The following 2 lines are called from within the dltfileindexerthread, which is not an owner of the plugins:

https://github.com/GENIVI/dlt-viewer/blob/master/src/dltfileindexerthread.cpp#L113
https://github.com/GENIVI/dlt-viewer/blob/master/src/dltfileindexerthread.cpp#L143

That leads to a race condition, if a plugin modifies its state in GUI thread simultaneously with these calls.

The way around, which I've applied to my plugin was:
- Make my plugin a QObject
- Use QMetaObject::invokeMethod in order to forward call to the corresponding thread

Please, either add comment to the mentioned method regarding in which thread they are called, or add implicit transfer to the correct thread before making a call.

alexmucde, 2019-08-13:
Yes you are right, this could be a problem. The Indexing thread was long time after the plugin interface definition.

Yes we should add some comments here.
The plugins must handle this use case with some mutex.

If you can provide a patch would be create.

svlad-90, 2019-10-22:
Hi @alexmucde, this issue can be closed as with the following commit I've already added the requested comments:

https://github.com/GENIVI/dlt-viewer/pull/49/commits/d877610ad194cb9f7ccb6e35927d14859d273fde#diff-2793c20d3915292d92240e590f628ea1R187

alexmucde, 2019-10-22:
Already fixed.

closed 2019-10-22 by alexmucde