COVESA/dlt-viewer #53: [2.19]Compiling with MSVC 2019 and Qt 5.13.1: conflicting of declaration of pid_t and missing linking library

dutchhuang, 2019-09-16:
When compiling with MSVC 2019 and Qt 5.13.1, there are 2 issues that I met:
1. When compiling, reported "error: conflicting types for 'pid_t' ... previous declaration of 'pid_t' was here..."
2. When linking, reported errors like "undefined reference to `__imp_setsockopt'".

dutchhuang, 2019-09-16:
Quick fixes for them:
1. comment the conflicting pid_t definition in the qdlt/dlt_types.h;
2. add '-lws2_32' to the 'LIBS' in the src/src.pro.

alexmucde, 2019-10-14:
Hello dutchhuang,

can you provide a pull request to fix this issue?
With this fix is this still working with Linux building and older MSVC versions?

    Alex

alexmucde, 2019-11-13:
@dutchhuang Any update in this issue?

dutchhuang, 2019-11-16:
Hello Alex, 

sorry for replying so late! I don't use the registered mailbox very frequently, so I wasn't aware of your comment on it.

 I didn't spend much time on investigating the root cause of this compiling issue, and all I do was just a dirty quick fix, so I'm afraid that I cannot answer your question very accurately. Somehow I think it is not difficult to locate the problem.

My gut feeling is, adding those definitely will affect compiling on the other platform. At least the ws2_32 library only exists on Windows platform I think. As for the pid_t, I think adding a namespace or using preprocessor directives can help on solving the issue.

At the last, I'd love to help on the fix, but to be honest I never had chance to go deep on this whole project, so I'm not sure whether I'm the best candidate on doing the pull request. If you insist I can take a try.

olaf-dreyer, 2019-11-22:
Hi All,

I fixed on my branch https://github.com/olaf-dreyer/dlt-viewer

1. with this commit 21ccc02a3fcddc49db8af8dd0256312fad765a66 (Fixing compile problem on mingw64_64)

and
2. with this commit 41176a254d2a19547ecb3b6959a874407a6fb71b (Fixing link problem on mingw64_64)

Because this is a problem on mingw64_64, too.
But I don't know, if my patches are compatible to other platforms.

Regards
Olaf

alexmucde, 2019-11-25:
@dutchhuang Applied patch from Olaf on master, please check if master now solves the issue.

alexmucde, 2019-12-06:
Please check and reopen if not solved yet

closed 2019-12-06 by alexmucde