COVESA/dlt-viewer #55: Try to connect to ECU on serial port "ttyUSB0"

Lawjune, 2019-09-29:
Hello,

I had installed this application by following the instruction of INSTALL.txt and been able to launch dlt-viewer application in Ubuntu 18.04. 

But when i tried to connect my ubuntu laptop to the ECU (which is able to publish dlt log to Windows laptop), the TTL232 connection could not establish. Can you kindly help me to figure out what kind of drivers needed to be installed? 

Thanks a lot in advanced!

alexmucde, 2019-10-14:
Hello Lawjune,

this should work in Ubuntu without any issues. Perhaps you try first getting your serial Port working with some other terminal tools. On some Linux platforms the port configuration does not work immediatly.  And when this is working you can try again with DLT Viewer.

Best regards

     Alex

alexmucde, 2019-11-13:
I think this problem is solved. Please reopen if this is still an issue.

alexmucde, 2019-11-15:
We will try to reproduce the issue.

alexmucde, 2019-11-15:
Duplicate of #65

Lawjune, 2019-11-20:
Hi Alex,

It`s my fault of setting the wrong connection configuration. Very sorry and it`s been working now.

Thank you so much!

closed 2019-11-20 by Lawjune