COVESA/dlt-viewer #56: DLT viewer jumps to line number instead of message with index x from "Search Results"

ssugiura, 2019-10-15:
DLT viewer jumps to unexpected lines with following scenario:
- Open lots of DLT logs.
- Click "Filter" -> "Sort by Time"
- Search something
- Double click a row in the Search Results
- DLT viewer does not jump to the message, but to the row of (Search Result)->Index

alexmucde, 2019-10-18:
I tried to reproduce this issue. I have loaded several files. Enabled Sort by Time. Then searched for a specific text and found four messages. When i double click one message the Table View jumps and hilights the correct message.
Can you please retry with lattest DLT Viewer version? We had some issues in the past.

ssugiura, 2019-10-29:
@alexmucde I tried with the latest DLT Viewer version and could confirm that it's solved. So I will close this issue.

closed 2019-10-29 by ssugiura