COVESA/dlt-viewer #63: Compatibility of dlt-viewer "v.2.19.0 STABLE" with Package Version "v.2.17.0 STABLE"

svlad-90, 2019-10-25:
Within my project I've switched to dlt-viewer v.2.19.0, while dlt-daemon on target is still v.2.17.0.

When I was using version v.2.18.0 of dlt-viewer I had no issues, while after update parsing seems to be broken. Most of messages are dropped. Indexing running to fast. It seems that it skips the data. I've found out that the following change is the reason:
https://github.com/GENIVI/dlt-viewer/commit/40e94c58bf7e21cf24d07d0927d46317a4aa3723

My question is: does above change have a counter-part change in dlt-daemon? In other words, was binary compatibility between daemon of version v.2.17.0 and viewer of version v.2.19.0 broken by purpose?

I can't switch my project to newer version of daemon, but still want to use version 2.19.0 of viewer. Any suggestions on how to achieve that? Will above commit be the only thing to revert in my local build of the viewer?

alexmucde, 2019-10-29:
There is no relationship between DLT Daemon and DLT Viewer versions. All should be comaptible as long the same DLT protocol specification is implemented. There is no incompatible change in the specification since the first release.

The change you are mentioning is an improvement of the indexing of DLT files. The problem here is that this change made things even worse, when the DLT files are corrupted, Thats why there was another change later after the v2.19.0 release "Further improvement and speed up indexing algorithm" ac7418d , which fixed this issue. So you have to apply the ac7418d patch to the v2.19.0 version.

alexmucde, 2019-11-13:
I hope this issue is solved, please reopen if not.

closed 2019-11-13 by alexmucde