COVESA/dlt-viewer #65: When the connection mode is serial port connection, the viewer cannot receive the message from the daemon.

Gonglja, 2019-10-31:
Hi, developer
I tried to use dlt-viewer on the ubuntu pc side, and open the dlt-daemon and log demo program on the pc side. The tcp network communication can receive the message from the dlt-daemon, but when I use the serial port, I can't receive the message. .
The hardware connection is as follows: pc side uses two usb to ttl module, tx connects rx, rx connects to tx, common ground.
Software configuration: dlt.cfg
RS232DeviceName = ttyUSB
Add RS232Baudrate = 115200  
already open,
The dlt-viewer is configured as a serial port connection, the port is ttyUSB1, and the baud rate is 115200.
[image]

[image]

Ps: Use the serial port debugging assistant to connect the two peripherals normally.

Gonglja, 2019-11-01:
Linux use ubuntu16.04

alexmucde, 2019-11-15:
@Gonglja Ubuntu 16.04 already reached end of life, but we will try to reproduce with Ubuntu 18.04.

Duplicate of #55

Gonglja, 2019-11-20:
ths, developer, very helpless, because the project was developed in ubuntu16.04, the customer is also using ubuntu16.04，all in ubuntu16.04...

alexmucde, 2019-11-22:
Looks like configuration issue, see also #55

closed 2019-11-22 by alexmucde