COVESA/dlt-viewer #66: [NonVerbose] Unknown argument type interpreted as string

vadrianc, 2019-11-13:
If an unknown argument type is used inside a frame found in a Fibex file, the `NonverbosePlugin::ParseFile(QString)` interprets it as a string:

`                      else
                      {
                          pdu->typeInfo = 0; // zero means a string (see below enum)
                      }`

when it should use the already defined `DltTypeInfoUnknown`:

`    //! The type definition of a DLT message argument.
    typedef enum { DltTypeInfoUnknown = -2, DltTypeInfoStrg = 0,DltTypeInfoBool,DltTypeInfoSInt,DltTypeInfoUInt, DltTypeInfoFloa, DltTypeInfoRawd, DltTypeInfoTrai, DltTypeInfoUtf8  } DltTypeInfoDef;`

The unknown PDU I defined and used in a frame: 

            <fx:PDU ID="PDU_21_4">
                <ho:SHORT-NAME>PDU_21_4</ho:SHORT-NAME>
                <fx:BYTE-LENGTH>0</fx:BYTE-LENGTH>
                <fx:PDU-TYPE>OTHER</fx:PDU-TYPE>
                <fx:SIGNAL-INSTANCES>
                    <fx:SIGNAL-INSTANCE ID="S_21_4">
                        <fx:SEQUENCE-NUMBER>0</fx:SEQUENCE-NUMBER>
                        <fx:SIGNAL-REF ID-REF="S_FOO2"/>
                    </fx:SIGNAL-INSTANCE>
                </fx:SIGNAL-INSTANCES>
            </fx:PDU>

alexmucde, 2019-11-15:
Yes you are right this is an issue. We have also to change the typeInfo variable to integer type.

@vadrianc Can you try this and provide a bug fix Pull Request?

alexmucde, 2020-03-13:
Is this still an issue?
Can you provide a solution for this?

alexmucde, 2020-04-01:
Any Update on this issue?

alexmucde, 2020-07-08:
No update, please reopen if still an issue.

closed 2020-07-08 by alexmucde