COVESA/dlt-viewer #67: Installing on windows 8.1

ThenewOne-p, 2019-11-13:
Hello to everybody,

i trying to install dlt viewer on my win8 64 bit.

can anybody explain how to do that? 

i am not a programmer and do not have skills but i want to learn.

thank you verry much.

regards

alexmucde, 2019-11-15:
Hello @ThenewOne-p 

please look at the INSTALL.txt. This should provide you some first information how to compile and install it.

First start with installing Qt SDK with Qt Creator and MSVC2015. Load the BuildDltViewer.pro file and compile and run the DLT Viewer with the Qt Creator.

Next you can try the Batch file build_sdk_windows_qt5_MSVC.bat. This will compile and copy all needed files to run the DLT Viewer without Qt Creator. Take care to set the correct pathes in the bat file first.

You do not need an installer to get the DLT Viewer running. Important is that the MSVC 2015 Rutime Environement is installed.

alexmucde, 2019-11-27:
I hope this issue is fixed, if not please reopen ticket.

You can also now find ready build zip files for Windows on Appvoyeur, e.g. here:

https://ci.appveyor.com/project/alexmucde/dlt-viewer/build/artifacts

closed 2019-11-27 by alexmucde