COVESA/dlt-viewer #68: [NonVerbose] Hexadecimal, binary and pointer arguments not being displayed

vadrianc, 2019-11-13:
If a hexidecimal, binary or pointer argument is sent in a DLT non-verbose packet, DLT Viewer does not display the value when applying the corresponding Fibex file.

               // HEX
		uint8_t hex8data = 0x1a;
		DLT_LOG_ID(con_exa3,DLT_LOG_INFO,DLT_EXA3_CON_EXA3_ID13,DLT_HEX8(hex8data));
		uint16_t hex16data = 0x1ad3;
		DLT_LOG_ID(con_exa3,DLT_LOG_INFO,DLT_EXA3_CON_EXA3_ID14,DLT_HEX16(hex16data));
	        uint32_t hex32data = 0x1abcd3e4;
		DLT_LOG_ID(con_exa3,DLT_LOG_INFO,DLT_EXA3_CON_EXA3_ID15,DLT_HEX32(hex32data));
		uint64_t hex64data = 0x17b4ddcf34eabb2a;
		DLT_LOG_ID(con_exa3,DLT_LOG_INFO,DLT_EXA3_CON_EXA3_ID16,DLT_HEX64(hex64data));

		// BIN
		uint8_t bin8data = 0xe2;
		DLT_LOG_ID(con_exa3,DLT_LOG_INFO,DLT_EXA3_CON_EXA3_ID17,DLT_BIN8(bin8data));
		uint16_t bin16data = 0x1234;
		DLT_LOG_ID(con_exa3,DLT_LOG_INFO,DLT_EXA3_CON_EXA3_ID18,DLT_BIN16(bin16data));

		// POINTER
		int *p = &num;
		DLT_LOG_ID(con_exa3,DLT_LOG_INFO,DLT_EXA3_CON_EXA3_ID19,DLT_PTR(num));

This related to the issue reported here https://github.com/GENIVI/dlt-viewer/issues/66, since the "S_HEX8", "S_HEX16", "S_HEX32", "S_HEX64", "S_BIN8", "S_BIN16" and "S_PTR" are not parsed by the non verbose plugin and the argument type defaults to 0, which means the values are interpreted as strings.

alexmucde, 2019-11-15:
You are right this is not supported so far.

@vadrianc Can you provide a Pull Request to provide the missing function?

olaf-dreyer, 2019-11-25:
According to the AUTOSAR standard 4.0, DiagnosticLogAndTrace.pdf, Table 7-24 Assembly of Type Info, these parameter types are not included, not defined.
Therefor the standard needs to be extended, before implementing something.
Or do I miss something?

alexmucde, 2019-11-27:
If this is an extension to the standard it hink this is no problem if we provide an extension in the source code. It is only important, that it will be compatible to the standard.

olaf-dreyer, 2019-11-27:
> WARNING: This e-mail originated from outside of Magna. Be cautious with links and attachments.

My idea is to not extend the standard.
The PDU would be transferred as UINT and the dlt-viewer will shows this as HEX, because of the definition in the FiBEX file.

alexmucde, 2019-11-27:
Ok then the Fibex types will be extended or based on the existent Fibex standard.

olaf-dreyer, 2019-11-27:
I would prefer the latter.
I have to sear, if we have that FiBEX standard.
Our could you send me the related informations?

Regards
Olaf

alexmucde, 2020-03-13:
Is this still an issue?
Can you provide a solution for this?

olaf-dreyer, 2020-03-13:
As far as I know, this is still an issue.
Due to project constraints, I wasn't able to to work on that.

alexmucde, 2020-04-01:
Any Update on this issue?

alexmucde, 2020-07-08:
No update, please reopen if still an issue.

closed 2020-07-08 by alexmucde