COVESA/dlt-viewer #72: "Silent mode" is flickering with gui window

noby256, 2019-11-21:
dlt_viewer with silent mode is still openning a window for a moment and closes it.

It is not bad, unless want to run a script for lots amount of dlt files for filtering.
Then it becomes an irriteted problem for you.

Alternative:
dlt_viewer could have cmd-line option for loading all files from a folder. Then you do not need to open a new dlt_viewer process per a log file.

I'm using
2.19.0, rev 733
Windows 10.

alexmucde, 2019-11-25:
@noby256 

As i understand you correctly, you want to convert all single files of a directory. Or do you want to load all files of a directory and then create a single export file of all opened files?

alexmucde, 2020-03-13:
Is this still an issue for you?

alexmucde, 2020-04-01:
No feedback anymore, please reopen, if this is still an issue.

closed 2020-04-01 by alexmucde