COVESA/dlt-viewer #83: [NonVerbose] Array of values in a single log message

sanyasinaidur, 2019-12-04:
How can I send an array of  values in a single log message for example an array of 100 values in DLT non-verbose message?

//Array of Values
uint8 u_index;
uint8 MyArray_1[100U];
uint16 MyArray_2[100U];

for(u_index = 0U; u_index < 100U; u_index++)
{
  MyArray_1[u_index ] = u_index + 1U;
  MyArray_2[u_index ] = u_index + 1U;
}
DLT_LOG_ID( myContext, DLT_LOG_INFO,  DLT_MSG_ID_IUC_TEMP,   DLT_ARAY8(&MyArray_1[0U]) );
DLT_LOG_ID( myContext, DLT_LOG_INFO,  DLT_MSG_ID_TDPU_EMP,  DLT_ARAY16(&MyArray_2[0U]);

alexmucde, 2020-03-13:
I think this is not supported in the current implementation. I think the only possibility is to send a memory buffer with DLT_ARG_RAW. If you need this feature please provide a pull request.

alexmucde, 2020-04-01:
Any update on this issue?

alexmucde, 2020-07-08:
No update, please reopen if still an issue.

closed 2020-07-08 by alexmucde