COVESA/dlt-viewer #84: ECU connect[Connection refused]

saitamo, 2020-01-04:
dlt-daemon version: 2.18.4 STABLE, Target IP: 192.168.1.10
dlt-viewer version: 2.19.0 UNSTABLE, PC IP: 192.168.1.100

The network on both sides can be Ping. I have an example program that circulates log output all the time. when I choose TCP connection, the dlt-viewer always show the connection refused. But I choose UDP connection, dlt-viewer connected the target device is successful, but displayed control type and dlt-viewer was refreshed every 5s. 
can you help me to analyse it. Thanks.
[image]

alexmucde, 2020-03-13:
I think this is a network issue on your side.
Please close ticket if it is already solved.

alexmucde, 2020-04-01:
No feedback anymore, please reopen, if this is still an issue.

closed 2020-04-01 by alexmucde