COVESA/dlt-viewer #85: dlt-receive command crashes when huge log load comes in

hkabra, 2020-01-07:
Hi,
I am using dlt-receive -f filename command to stream the logs but it crashes when a log which is quite huge in size comes in.
Unfortunately the log of interest is the huge log.

The dlt viewer works fine but when I get the logs using dlt-receive, it crashes.

ssugiura, 2020-01-08:
@hkabra dlt-receive is developed in dlt-daemon repository, not in dlt-viewer. So could you please re-raise the issue in dlt-daemon git instead with more detailed information, e.g. what is the signal raised, how much logs outputted?
BTW, -f is an option to filter messages. If you expect to output messages in dlt log file, you need to use -o.
Further discussions should be continued in dlt-daemon git.

hkabra, 2020-01-08:
Thanks, I posted it there.
Yeah I used -f to filter messages but the problem, I am facing is that a single log line was so huge that it crashes the streaming of logs.

ssugiura, 2020-01-16:
Will discuss in https://github.com/GENIVI/dlt-daemon/issues/201.

closed 2020-01-16 by ssugiura