COVESA/dlt-viewer #86: incorrect scaling on high dpi display

markus-i, 2020-01-20:
Easily explained: I am on a high dpi display (3840x2160) in Windows 10. Windows recommends a display scaling of 150% (which is more of a minimum). The result in dlt-viewer is horrible: the text is scaled, but the bounding boxes of the tables aren't. Manual/automatic adjustment of the width is possible, but not of the height, with the result that the characters are squeezed into the lines, and characters going below the baseline (like q, g or y) are cut off.

DLT version is a customized (for one of our plugins) version of 2.19.0, built with QT 5.5.1

Question: does DLT (and QT) support high dpi displays and if yes, any suggestions on what we could do to get a reasonable display again?

[image]

alexmucde, 2020-03-12:
Hi Markus,

i tried this at my UHD monitor at home with following build:

https://ci.appveyor.com/project/alexmucde/dlt-viewer-2/build/artifacts

MSVC2017 Qt5.13.2

I used 150% upscaling. I get a different result than you.

Only the DLT message table view looks ugly, it tries to upscale the Fond, perhaps we can try to change the Fond, perhaps also configurable in the seetings.

The search table view looks ok.

Alex

alexmucde, 2020-03-13:
Pushed to master new feature to set the Font and the height of the table sections.

You can try the following build.

https://ci.appveyor.com/project/alexmucde/dlt-viewer-2/build/artifacts

Tried this on my UHD monitor and works fine.

Please reopen if this still does not fix your issue.

closed 2020-03-13 by alexmucde

markus-i, 2020-03-13:
Thank you. I'll check it.