COVESA/dlt-viewer #88: getMsg: Index 185984 is out of range 1243 ..\..\dlt-viewer\qdlt\qdltfile.cpp line 554

SvenLuebke, 2020-01-27:
Hallo!

This error is reported endlessly (within a fast loop) in the console after "working" with Dlt-Viewer. Unfortunately I didn't spot the root cause action yet, as I was analyzing our issues.

alexmucde, 2020-03-12:
Hello Sven,

i tried to reproduce, but without success.
Can you try the lattest version of DLT Viewer on the master?
Any progress in this topic from your side?

    Alex

alexmucde, 2020-04-01:
Any update on this issue?

SvenLuebke, 2020-04-01:
Hi Alex!

Sorry for this late answer. I tried to analyze this issue, but it wasn't easy to find (at least for me as I don't the sources that good). I tried to update to Qt release 5.14.1 (hoping, that it magically disappears) first and updated to latest master, but that didn't fix it.

I just noticed that this message won't get reported if I click on the first (sometimes 3, sometimes 10000) search results. Could you try to reproduce again (applying a filter, then search for something, that will occur at least 30000 times) and then try to select the last entry in the search table. If that doesn't reproduce it, I guess it might be happening due to a malformed Dlt message?!

Unfortunately I'm not allowed to distribute the dlt logs. I'll try to generate some simple ones which I can send to you in case you can't reproduce this issue.

Thank you and BR,
Sven

SvenLuebke, 2020-04-02:
Sorry...this issue didn't let me sleep, but now - I think - I have an explanation for most of the things.

I'll currently assume, that you get this "issue" as well...but you just don't notice it. The reason why I am noticing it is, that I linked "*.dlt" and "*.dlp" to dlt-viewer.exe (which "speeds up" opening of Dlt files) and in this mode (probably my setting might be wrong) a console window is opened in parallel, which shows some status information. Until now, this window didn't disturb me, so I left it appearing. In #98 I noticed the first time, that this console window also needs more CPU time (than I expected)...at least if a lot of messages will appear and that's the case with this issue. Running into this "issue" with an opened console window will make dlt-viewer nearly unusable, as all the messages will be printed. If I just double-click dlt-viewer.exe, I don't get any console window so the speed is back to normal again.

Now the only question for me is: Is that state "Index xxx out of range" probably harmful or is it clear why dlt-viewer runs into this situation? It did sound wrong to me...but if that is normal I will deactivate this qDebug() message and then we can close this issue.

BTW: You can also see these messages in Dbgview from the SysInternalsSuite. Or did you probably define out all qDebug() stuff, so it will be quiet. I'm building dlt-viewer via QtCreator.

Thanks for your support!

Best regards,
Sven

alexmucde, 2020-04-07:
I can see the qDebug() output only, when i am compiling and starting DLTViewer from QtCreator. In normal Release build qDebug should be disabled and influence not the performance.

Good hint regarding the tablemodel.cpp. It could be really possible that lastSarchIndex is not a valid Index in the search table model, which could result in a out of range Debug message. I am currently do not find the "real" problem here. I think there needs more debugging. If you find the real issue please let me know.

alexmucde, 2020-07-08:
Any news on this topic? Any pull request?
I think if it is only a qDebug message thats not an issue.
If not an issue i will close this ticket.

SvenLuebke, 2020-08-26:
Hi Alex!

Sorry for the long delay. I tried to debug this issue a little bit, but I know only a few state variables of this application.

If you double click in search dialog, the following function is executed
https://github.com/GENIVI/dlt-viewer/blob/e546fbb3d934f1a039e7704c2b5075c3e4d398f3/src/searchdialog.cpp#L191
This sets tablemodel internal variable lastSearchIndex via
https://github.com/GENIVI/dlt-viewer/blob/e546fbb3d934f1a039e7704c2b5075c3e4d398f3/src/tablemodel.h#L62
so it is set to the msg index of the clicked msg shown in searchdialog, thus NOT equal to -1. If you apply a (effective and working) filter now, the messages in the tableview will get reduced...let's say to 10, but the variable lastSearchIndex is still the unchanged value...let's say 100, which is not valid anymore. If Qt decides to update the table, it will call
https://github.com/GENIVI/dlt-viewer/blob/e546fbb3d934f1a039e7704c2b5075c3e4d398f3/src/tablemodel.cpp#L81
which leads to the execution of
https://github.com/GENIVI/dlt-viewer/blob/e546fbb3d934f1a039e7704c2b5075c3e4d398f3/src/tablemodel.cpp#L275
and then to this output. So the filtered messages are probably even 0, but the lastSearchIndex is still at a higher value.

As lastSearchIndex most probably can't be translated to this filtered value (the result of filter and search might not be related to each other) I would set lastSearchIndex to -1 again in
https://github.com/GENIVI/dlt-viewer/blob/e546fbb3d934f1a039e7704c2b5075c3e4d398f3/src/tablemodel.cpp#L396
I tested this and the output doesn't occur anymore! The benefit of this change would be a slightly better performance and no output while debugging other stuff.

What do you think?

BR,
Sven

alexmucde, 2020-09-11:
Hello Sven,

yes this fix sounds resonable. Please provide a PR for this.

Thanks

    Alex

alexmucde, 2020-12-11:
I have reworked the color handling of the table model. Perhaps the issue is not there anymore.
But to be sure i will add seeting the lastSearchIndex to -1 into the modelChanged routine.

closed 2020-12-17 by alexmucde