COVESA/dlt-viewer #89: Issue with UDP connectio.

spiderman07, 2020-02-03:
When creating a new connection in DLT Viewer, UTP type one with Multicast on network device enabled and ip provided and correct network interface created, i get error on DLT Viewer saying Binding failed with this error:  QAbstractSocket::SocketAddressNotAvailableError

The device is connected to that network and is responding to ping. Also i can see wireshark packets comming from daemon from target.

In order to fix it i changed on the code in src/mainwindow.cpp line 3112:

bindstate = ecuitem->socket->bind(QHostAddress(ecuitem->getmcastIP()), ecuitem->getIpport(),QUdpSocket::ShareAddress );

To this:

bindstate = ecuitem->socket->bind(ecuitem->getIpport(),QUdpSocket::ShareAddress );

Basically i removed the ip adress used for binding and let it bind on the port. Then It nicely joins the multicast group and logs do appear.

Is this a bug in code? Anyone tested this UDP connection until now?

PS: let me know if you need any more info.

Regards,
Razvan

alexmucde, 2020-03-12:
Is this issue in Windows or Linux?

spiderman07, 2020-03-12:
Hi.

Is this issue in Windows or Linux? -> Windows
Do you have several Ethernet interfaces on your system? -> Yes i have more. 2 adapters +1 from virtual box
Did you seleected the right Multicast Interface? yes i did select the correct one. i have tried all of them.

olaf-dreyer, 2020-03-13:
There is a bug in the Qt code (as far as I know).
In my own product, I needed to implement a workaround:

        if(!m_strLocalAddress.compare("0.0.0.0") || m_strLocalAddress.isEmpty())
            bBound = m_pUdpRxSocket->bind(QHostAddress::Any, m_iLocalPort, QUdpSocket::ShareAddress);
        else
            bBound = m_pUdpRxSocket->bind(QHostAddress(m_strLocalAddress), m_iLocalPort, QUdpSocket::ShareAddress);

        quint32 group = QHostAddress(m_strRemoteAddress).toIPv4Address();

        if(bBound && (group >= 0xE0000000) && (group <0xF0000000))
        {
//            qDebug() << "ComDeviceSocket::open: searching interface for multicasting";

            int i = -1;

            QList<QNetworkInterface> lIF = QNetworkInterface::allInterfaces();
            for(i = 0; i < lIF.size(); i++)
            {
//                qDebug() << "ComDeviceSocket::open: interface" << i << ":" << lIF.at(i).humanReadableName();

                int a = -1;

                QList<QNetworkAddressEntry> lAddr = lIF.at(i).addressEntries();
                for(a = 0; a < lAddr.size(); a++)
                {
                    QNetworkAddressEntry e = lAddr.at(a);
                    QHostAddress addr = e.ip();

//                    qDebug() << "ComDeviceSocket::open: interface" << i << ", addr" << a << ":" << addr.toString();

                    if(addr == QHostAddress(m_strLocalAddress) || !m_strLocalAddress.compare("0.0.0.0") || m_strLocalAddress.isEmpty())
                    {
                        qDebug() << "ComDeviceSocket::open: join group on interface" << i << ", addr" << a << ":" << addr.toString();

                        m_pUdpRxSocket->joinMulticastGroup(QHostAddress(m_strRemoteAddress), lIF.at(i));
                        //m_pUdpTxSocket->joinMulticastGroup(QHostAddress(m_strRemoteAddress), lIF.at(i));
                    }
                }
            }
        }

        if(!bBound)
            qDebug() << "ComDeviceSocket::open: Bind problem, unable to bind to" << m_strLocalAddress;
        else
            qDebug() << "ComDeviceSocket::open: bound to" << m_pUdpRxSocket->localAddress() << ":" << m_pUdpRxSocket->localPort();

alexmucde, 2020-04-01:
Can anyone provide a patch/Pull Request for this issue?
Or is it solved in Qt with any release of Qt?

sebastianlipponer, 2020-04-21:
I am able to connect via UDP with Multicast on network interface enabled using 457cef969 based on QT 5.14.1.

alexmucde, 2020-07-08:
Looks like it is solved, please reopen if needed.

closed 2020-07-08 by alexmucde