COVESA/dlt-viewer #91: DLT-Viewer is frozen while doing "Save As..."

thanhbnq, 2020-02-20:
Hello dlt-viewer experts,

At the latest commit(18f1ec9601: Fix filter item...), there is an issue and it can be easily to reproduce as following scenario:
1. Board is on but dlt-daemon is not running yet.
2. Open dlt-viewer on Ubuntu
3. Configure ECU and TCP connections
4. Click "Connect all ECU's or create new one". In console terminal, there are logs:

> "Socket connection error "Connection refused" for <IP_address> on 3490".

5. Try to use "Save as..." to dlt file
6. Now everything is frozen. Cannot do anything with dlt-viewer unless terminate it. In console terminal, there are log like:

> Socket connection error "Connection refused" for <IP_address> on 3490
> ERROR bytesRcvd <= 0 in  3540 ../../dlt-viewer/src/mainwindow.cpp
> QNativeSocketEngine::bytesAvailable() was called in QAbstractSocket::UnconnectedState

There won't be any issue if:

- dlt-daemon is running and dlt-viewer can connect to it successfully. "Save As..." is working fine.
- Or Click "Disconnect all connected ECU's" before performing "Save As..."

Could you please check it!?

Thank you in advance!
Thanh

thanhbnq, 2020-03-11:
@alexmucde Could you please give your opinion on this case as well !?

alexmucde, 2020-03-12:
I tried to reproduce in Windows, but without success. I see the same Debug output as you.

thanhbnq, 2020-03-26:
@alexmucde sorry for my late response!
I was testing with Ubuntu 16.04. I will try if I can give pull request for this issue.

alexmucde, 2020-04-01:
Any update on this issue?

thanhbnq, 2020-04-01:
@alexmucde I am sorry that I cannot give any try for the fix of this issue in short time. It will be really great appreciation if you (with great experience and knowledge) could help to resolve it better than me!

thanhbnq, 2020-04-28:
Hello @alexmucde ,
After checking with a clean build again several times, I realize that I myself made a mistake while compiling dlt-viewer in Ubuntu.

From the "Configure Project" of QT Creator, there are two suggestions for me, "Desktop" and "Desktop Qt 5.x.x GCC 64 bit", then "Desktop" is my default choice for project. It must be "Qt GCC" one, this is my mistake.

Then now I verify the reported scenario and there is no any issues observed.

So I would like to close this issue as no fix required.

Thanks for your time so far and sorry for my bad mistake!

Thanh

closed 2020-04-28 by thanhbnq