COVESA/dlt-viewer #92: Problem: Serial connection with "Sync to Serial Header when receiving"

thanhbnq, 2020-03-10:
Hello dlt-viewer experts,

Please kindly check the following scenario:
1. tty-console.service is still active
2. start dlt-viewer
3. connect to ecu via serial configuration.
4. stop tty-console.service
5. run dlt-daemon, run dlt-example-user (from GENIVI/dlt-daemon) to send messages.
6. Nothing in dlt-viewer.

I try to analyze and I think the issue is about the "Sync to Serial Header when receiving" (Under "Serial" Tab ) is not updated correctly. 

There is a commit (long time ago) [Add UDP Unicast / Multicast reception: ](https://github.com/GENIVI/dlt-viewer/commit/896b1ad4245b57c4c2da60bf252fa98ca25e949b), I think it could be involved this issue. In that commit, the syncSerialHeaderSerial is not updated into EcuItem:
`-    item->serialcon.setSyncSerialHeader(this->syncSerialHeaderSerial());`

So it makes the parser do not try to sync the serial header until the proper header is found out.
I try to add the following line into "EcuDialog::setDialogToEcuItem":
`+   item->setSyncSerialHeaderSerial(item->getSyncSerialHeaderSerial());`

And it seems to be working fine.

So I would like to ask your consideration on this case!

Thank you in advance!
Thanh

alexmucde, 2020-03-10:
Thanks for the feedback. Looks like a real problem. I will provide a fix.

alexmucde, 2020-03-10:
Your fix proposal is not correct. The right fix is:

    item->setSyncSerialHeaderSerial(this->ui->checkBoxSyncToSerialHeaderSerial->checkState()==Qt::Checked);

Also the other paramter sendSerialHeader is not handled correctly. Will fix this also.

thanhbnq, 2020-03-11:
Thank you @alexmucde !

alexmucde, 2020-03-11:
Fixed and pushed to master.

closed 2020-03-11 by alexmucde