COVESA/dlt-viewer #99: Filter with regular expression not workinh

ZiadRhouma, 2020-04-05:
if I set a payload filter with regular expressions
, then no messages are shown in the DLT viewer, even though the messages already exist.

[image]
[image]
[image]

alexmucde, 2020-04-06:
We are using the QRegularExpression class from Qt libary. According to the Qt doumentation the regular expression feature CaretAtZero you want to use is not supported:

https://doc.qt.io/qt-5/qregularexpression.html#caret-modes

bazurbat, 2020-06-02:
Hi! This is simply a bug, I've just made a PR (#102) which fixes the problem. Please check it out. It looks like it's doing what it is supposed to do with the caret for the logs which I have at least.

alexmucde, 2020-07-08:
#102 will solve this issue

closed 2020-07-24 by alexmucde