COVESA/dlt-viewer #100: Error collect2.exe during compilation

fxkdev, 2020-04-08:
Hi,

I tried to build dlt-viewer 2.19.0 with Qt Creator (Qt5.12.4) and got the following error (that did not happen with dlt-viewer 2.18.0 and the same project-configuration).

[image]

Do you have any idea what it is related to? Is my project bad configured or can it be a linked to a missing library?

Thank you for your support

alexmucde, 2020-04-08:
Hi @fxkdev. We currently do not support MinGW builds. Please switch to MSVC build on Windows platform.

alexmucde, 2020-07-08:
No official support for MinGW. please provide pull request if still needed.

closed 2020-07-08 by alexmucde