COVESA/dlt-viewer #101: Filters don't apply

Wasted90Youth, 2020-05-13:
Hi.

My program  uses 
D:\\DLTViewer-2.19.0-WORKING-5.8-r711_x64\\dlt_viewer -f D:\\DLTViewer-2.19.0-WORKING-5.8-r711_x64\\filters\\positive_DCS.dlf -csv -c

command which is working fine on different PCs. But one of them just creates a .csv where the filters doesn't seem to apply, which leads to bigger .csvs than .dlts

I already copied the whole DLTViewer folder from a PC where it's working to the one that doesn't work. But same procedure.

Is there anything I have to delete first to make it work?

johumuc, 2020-06-16:
Just a guess, 
did you check your config.ini  --> filtersEnabled= Settings.
Maybe you just have to start dlt-viewer GUI and set the checkbof for filters Enabled.

alexmucde, 2020-07-08:
Yes check your config.ini file in your personal settings directory. Perhaps there is a wrong configuration. Best delete it or copy it from another PC. Can we close this issue?

Wasted90Youth, 2020-07-08:
Well it's still the same issue. But I dont need it now. Thanks for your comments

closed 2020-07-08 by Wasted90Youth