COVESA/dlt-viewer #107: [Question] DLT-Message-Analyzer plugin

svlad-90, 2020-07-04:
Hi maintainers.

I've created this issue to mention, that there exist is a huge plugin for dlt-viewer:
https://github.com/svlad-90/DLT-Message-Analyzer
It extends the dlt-viewer's functionality quite a lot.
Please, check, what it does.

**=> If it worth it, I propose to mention it somewhere in the area of the dlt-viewer's repository.**

I'm currently using this plugin in one of the projects at work and plan to maintain it for a while. At least until dlt-viewer is the main trace-analysis tool that is used around in automotive.

**=> Please, let me know, whether you are not against mentioning the plugin, as I'm searching the ways to increase its audience. Still, I'll definitely understand if you'll reject such an idea.**

alexmucde, 2020-07-28:
I tried it, but i do not know how to use it. Can you explain in more detail what the plugin is used for and how it is used? Perhaps some simple examples will be useful.

svlad-90, 2020-07-28:
Hi @alexmucde, the installation guide is located here:
https://github.com/svlad-90/DLT-Message-Analyzer/blob/master/md/installation_guide/installation_guide.md
Use it to build the plugin as part of the dlt-viewer.
Then enable and show it in the dlt-viewer.

Plugin looks like this:
[image]

The general purpose of it is to use it instead of the usual search.

Use it in the following way:
1. Open some dlt-file or start the session with some target system
2. Enter your regex into the plugin's input field
3. Press analyze button
4. Analyze the results
That's the most simple scenario. The plugin supports much more.

To get familiar with the feature-set of the plugin check its README.md:
https://github.com/svlad-90/DLT-Message-Analyzer/blob/master/README.md

Some sections, which describe the main features of the plugin:
https://github.com/svlad-90/DLT-Message-Analyzer/blob/master/md/search/search.md
https://github.com/svlad-90/DLT-Message-Analyzer/blob/master/md/patterns_view/patterns_view.md
https://github.com/svlad-90/DLT-Message-Analyzer/blob/master/md/regex_name_scripting/regex_name_scripting.md
https://github.com/svlad-90/DLT-Message-Analyzer/blob/master/md/in_ram_cache/in_ram_cache.md
https://github.com/svlad-90/DLT-Message-Analyzer/blob/master/md/grouped_view/grouped_view.md

svlad-90, 2020-08-05:
Hi, @alexmucde. I've recently added support of PlantUML to DLT-Message-Analyzer. So now it is possible to generate UML sequence diagrams out of the logs on the fly.

More details here:
https://github.com/svlad-90/DLT-Message-Analyzer/blob/master/md/plant_uml/plant_uml.md

svlad-90, 2020-11-29:
@alexmucde, a kind reminder. Did you get a chance to test the plugin?

alexmucde, 2020-11-30:
Yes i tried shortly your plugin, but i did not know the exact purpose. I added a refernece in the documentation.

closed 2020-11-30 by alexmucde

svlad-90, 2020-11-30:
Hi @alexmucde, 

_I added a refernece in the documentation._

Thank you very much for fulfilling my request! That motivates me to continue the support of the DLT-Message-Analyzer! 

----

_Yes i tried shortly your plugin, but i did not know the exact purpose._ 

Hopefully, after my additional feedback in previous messages, the purpose of the plugin is clear for you. In case if still not - the page of the plugin, which you've already mentioned in the  dlt-viewer's README - https://github.com/svlad-90/DLT-Message-Analyzer - itself contains an extensive README section, which gives a precise overview of all the supported features. In general, it is a replacement for the dlt-viewer's search. It works faster, allows to highlight parts of found strings with different colors, supports conversion of the logs into the plantuml diagrams, allows to search within the range of the messages, provides support of regex-name-scripting, etc. 

If you want to know more - get familiar with README section. Otherwise - the contribution you've made is already enough, as the link to the plugin is sufficient to get all the information about it.

----

This issue can be closed. Thank you once again.