COVESA/dlt-viewer #115: Issue displaying DLT message containing other DLT messages

andreirusu96, 2020-08-31:
Hello!

I am currently using the Filetransfer feature of Genivi DLT to transfer a buffer containing a bunch of DLT logs (complete DLT logs containing the "DLT0x01" header). As you know, the Filetransfer basically sends DLT logs to DLT Viewer with the contents of the file as DLT_RAW(rawData, dataSize). However, there is an issue inside DLT Viewer that causes the DLT Filetransfer log (the "FLDA ... FLDA" one) to show as "CORRUPTED MESSAGE!!!" when it contains more than 2 DLT logs within itself. Please see below screenshot:

[image]

As you can see, the messages inside the payload are being read as actual DLT messages. How it should show up:

[image]

**Observations from testing:**

- At least 2 logs need to be contained within the "FLDA ... FLDA" Log for the corruption to happen (haven't managed to check out why this issue doesn't occur only one DLT message inside the payload)
- There was this commit that fixed receiving messages containing the DLT header in the payload ("DLT0x01") : https://github.com/GENIVI/dlt-viewer/commit/40e94c58bf7e21cf24d07d0927d46317a4aa3723 This did indeed fix the issue, however, this specific edge case was missed (see below).

**Observations from debugging:**

1. The "CORRUPTED MESSAGE!!!" error appears only when the "FLDA ... FLDA" log is the last log read in the "buf" variable:  
`buf = files[numFile]->infile.read(READ_BUF_SZ);`(line 204, file "qdltfile.cpp")

- In this case, the following would be true at the end of the method call: `file_size == next_message_pos` , so the following method will be called: `files[numFile]->indexAll.append(current_message_pos);` (line 232, file "qdltfile.cpp")
- The aforementioned commit fixed the handling of such messages when the "FLDA ... FLDA" would be placed in begging/middle of the "buf" buffer, but this specific case was missed. 
2. When such a message is the last read in the "buf" buffer, the next call to `QDltFile::updateIndex()` (line 156, file "qdltfile.cpp") would take the last message's position in the buffer and continue looking for messages from that `position + 4`, skipping the "FLDA ... FLDA" message's DLT header: `pos = (*const_indexAll)[files[numFile]->indexAll.size()-1] + 4;` (line 179, file "qdltfile.cpp"). _From my point of view, this is the root cause and the method that needs fixing. However, I don't have experience with the DLT Viewer codebase..._
- Then, the next detected DLT header would be inside the payload, since only the FLDA log's header would get skipped.
- Finally, DLT Viewer reads each message's buffer by calculating the difference between the positions of message 'i + 1' and 'i', as can be seen here: `long int cal_index = const_file->indexAll[index+1] - positionForIndex;` (line 509, file "qdltfile.cpp") and `buf = file->infile.read(cal_index);` (line 513, file "qdltfile.cpp").
- This would be OK for any other message that does not contain a DLT header, but not for this one. The total message length should be extracted and the entire message skipped, so no DLT header from the payload of the message should be encountered.

**Testing the issue:**

- Check the following gists for test code: [Test Code 1](https://gist.github.com/andreirusu96/18010c1913239599b984bf3d17ec2d30) and [Test Code 2](https://gist.github.com/andreirusu96/ff2128419ee5fb14239e1241c3b33e07)
- For easier debugging: use the large message and set a breakpoint on _line 228, file "qdltfile.cpp"_ with the following condition: `message_length > 1000`
- Do not set a breakpoint for each read message as this could affect the "real" (no break) behavior of the reading of the buffer  
- Try sending the message multiple times, as not always the message will be on the end of the "buf" buffer (DLT Viewer will then display the message OK)
- Reduce the logging to DLT Viewer -> On my machine, there aren't many logs sent at once, but I suppose having many would reduce the chances of having the example DLT log at the end of the buffer.

Please let me know if there's any other information needed!

Thanks a lot!

alexmucde, 2020-11-13:
Can you provide a DLT File which contains a DLT message header in the payload?
Can you describe the use case when the problem occurs?

andreirusu96, 2020-12-03:
Hello, @alexmucde!

The use case is as follows: we use the Filetransfer feature of DLT to transfer a DLT file containing DLT logs. As you may already know Filetransfer splits the file in smaller chunks, so most of the time, inside a Filetransfer chunk, one of the transferred logs inside the file will get cut. This will cause problems, since the current DLT logs parsing algorithm does not take in consideration the contents of the Filetransfer Data logs. The algorithm will parse the contents of the transferred chunks and the index will become corrupted, since it will reach the cut message (considering it a normal message, since it also has a DLT message header.)

Unfortunately, I cannot provide a file, since this issue only happens for 'online' logs (please check the initial post for more details). However, I have included in the first post a DLT message example, which will cause the error to appear when sent using the DLT API. 

Also, it is not necessary to have a special case for Filetransfer chunks when parsing received DLT logs. As explained in the original comment, this is an edge case that was not taken in consideration when dealing with DLT logs containing other DLT logs inside the their payload. Please check the linked PR which has the fix for this issue.

alexmucde, 2020-12-18:
Sorry for being late. I had a deeper look into your patch.

There are three places where parsing DLT messages is done:
- qdltfile
- dltfileindexer.cpp
- qdltconnection.cpp

When loading DLT Files the first time dltfileindexer.cpp is used, that is what already was fixed.
Qdltconnection.cpp here we could have an additonal issue, but this only used for parsing serial headers, which are only used for serial connections.
You are right qdltconnection.cpp is used for updating the index when new messages arrived in the online use case.
Your solution looks good and i think i can integrated it.

Thanks for your help.

closed 2020-12-18 by alexmucde