COVESA/dlt-viewer #116: [Bug] no output after connection

artur522y, 2020-09-09:
Dlt_viewer info:
Package Version : 2.20.2 testversion
Qt Version: 5.12.5
Compiler: gcc
Compiler Version: 7.5.0

Precondition:
remote dlt server via tcp/ip

Error path:
connect to the host

Misbehavior:
empty output table, despite connection indicator is green

Expected behavior:
flow of logs/traces in the output table

Recovery path:
disable or enable filters and press "apply configuration", then all traces you missed will appear on the log stream will flow as usual.

alexmucde, 2020-09-11:
Need more information to reproduce the problem.

hkjin81, 2020-09-28:
Qt 5.15.1
mingw81_64
Windows 10
tcp/ip connection

I'm using tcp/ip connection in daily work and I couldn't reproduce the problem with the latest source code(2.20.2 test version).

4og, 2020-10-02:
I experience the same issue on Ubuntu 20.04 when connecting over TCP to a remote ECU. The output table stays empty; however, the messages are received (Recv. number in the status bar is increasing, and the search works). Qt 5.12.8, GCC 9.3.0

After doing the git bisect, it seems the issue was introduced in 53e3f541e90391676527da42dee86ff3e745eed1

hkjin81, 2020-10-22:
Same issue occured in Ubuntu 20.04 over TCP connection.(Qt 5.15.1, GCC 9.3.0)
After apply configuration, it starts to display as above comment but if I clear the log it doesn't display again.

alexmucde, 2020-10-22:
Thanks for the feedback, we will have a look.

alexmucde, 2020-11-09:
This should be fixed now. Return value was wrongly changed in 53e3f54.

closed 2020-11-09 by alexmucde