COVESA/dlt-viewer #118: View doesn't follow current cursor position in Windows platform.

hkjin81, 2020-09-28:
Precondition:
Version 2.18, 2.19, 2.20, 2.20.1
Qt 5.15
Windows 10

Steps:
1. Open a dlt log file.
2. Select one line by mouse.
3. Press down arrow key by keyboard until view scroll happen.

Expected result:
View scrolled automatically to keep current line cursor visible.
(Linux version is working as expected)

Actual result:
View doesn't follow the cursor and the cursor disappeared from current view.

Etc:
This problem make me to lose current scroll position after "Apply Configuration" clicked.
It goes to the top so I have to remember the line before "Apply Configuration" and scroll to the line by myself.
It's very annoying.

Please feel free to ask me further information.

alexmucde, 2020-11-13:
Will this issue be fixed with #120

alexmucde, 2020-11-19:
This issue only happens, when column "index" is disabled.

closed 2020-11-19 by alexmucde