COVESA/dlt-viewer #124: After applying filter rows with same time are sorted in descending order

ChronoTempus, 2020-11-05:
[image]
I have this issue that whenever I apply a filter and for any group of rows there is same Time then rows will be sorted in descending order (in image can be seen that Index and Count are also in descending order).
Even if I try to sort by timestamp (which is not always possible if I have multiple life cycles in same dlt) if time is same rows will be in descending order
[image]

I tried to find some settings that control this or something in source code itself, but didn't manage to fix this.

OXINARF, 2020-11-06:
I was coming to report something very similar, but I don't have the issue when applying only filters, but only when "Sort by time" or "Sort by timestamp" is also enabled. @ChronoTempus can you confirm if that's your case too?

The problem that triggers my issue is with how Qt manages the QMultiMap entries with the same key: when you insert an entry, if there is an existing one with the same key, the new one will **precede** the existing one. Since the log is iterated on increasing index order, the later messages with the same time/timestamp will end up being first on the log.

@alexmucde I would suggest that the DltFileIndexerKey also got the index so that if time/timestamp is exactly the same, then they are ordered by the index. I can do a PR for that, if you want.

ChronoTempus, 2020-11-06:
Can confirm if both checkboxes are disabled log is sorted correctly.

OXINARF, 2020-11-06:
Ok, so it's the same issue. @ChronoTempus maybe change the title to something like "Sorting by time/timestamp is in the wrong order"?

alexmucde, 2020-11-13:
Do you have any idea how to solve this issue? It would be even better if you can provide a pull request?

alexmucde, 2020-11-19:
PR integrated.

PR #131 (2020-11-19): Sort by time/timestamp keeps index order
For messages that have the exact same time/timestamp, we keep the same order for the messages by using their index.

Fixes #124

**This has not been tested**. @alexmucde if you can help with a test build it would be great.
commit 646be372 (2020-11-19): Sort by time/timestamp keeps index order

For messages that have the exact same time/timestamp, we keep the same order for the messages by using their index.

Fixes #124
closed 2020-11-19 by alexmucde