COVESA/dlt-viewer #141: macOS icon missing

hannesa2, 2020-11-23:
Hi, first of all: Thank you to make it public !

I build it with macOS and it builds and runs, but the icon is not available

[image]

Sorry, I've to less knowledge with `qt` to fix it

alexmucde, 2020-12-01:
Hi, i am not familiar with macOS. I have found the following instruction:

https://doc.qt.io/archives/qt-5.11/appicon.html

If you use qmake you need to do the following:

At first you need an icns file, which should be already in \src\icon .
Then you have to add to the src/src.pro file.
ICON = icon/icon.icns

If you are using cmake it will be more complicate, please see the instructions.

Please try this if it works you or me can create a pull request.

hannesa2, 2020-12-01:
To give anyone the chance to follow, I made a non-working pull request https://github.com/GENIVI/dlt-viewer/pull/149

alexmucde, 2020-12-17:
@hannesa2 so this is not fixed yet?

closed 2020-12-17 by alexmucde