COVESA/dlt-viewer #142: macOS Apple silicon M1 cpu build error

hannesa2, 2020-11-23:
During build on macOS M1 cpu I observe this, and I've no clue to solve this with `qt`
```
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [bin/dlt-parser] Error 1
make[1]: *** [parser/CMakeFiles/dlt-parser.dir/all] Error 2
make: *** [all] Error 2
```
Maybe someone knows how to solve it ?

alexmucde, 2020-12-01:
Sorry, i am not familiar with MacOS. Because someone working with MacOS can help.
I am wondering why you want to compile dlt-parser. This is an extra tool, which you do not need for DLT Viewer.

hannesa2, 2020-12-01:
I simply used the travis script `.ci/travis/build.sh`

alexmucde, 2020-12-17:
@hannesa2 You find a solution to solve this issue? Can we close?

hannesa2, 2020-12-17:
I'm able to build, but this particular issue is still open

Elastino, 2021-01-09:
Leaving a note here for the future reference.

The build script in `.ci/travis/build.sh`  assumes that Qt will live unde `/usr/local/opt/lib` which isn't the default location with the latest Brew installation. 

We can change the script to point out new Brew's folder such as `/opt/homebrew/....`
However, still, it won't allow us to execute App Package due to new policy from Apple on unsigned(developer's sign) App to be launched in M1 silicon, presumably protecting existing iOS Apps being stolen.

### Option 1 - Highly discouraged 
Disable **SIP** via command `csrutil disable` from Recovery Terminal. 
This will allow to launch DLT Viewer without code-sign but it will prevent launch of existing iOS apps.

### Option 2 - run as it is
If an artifact is built with `.ci/travis/build.sh`, the artifact in `build/release/` will continue to work. 
However, it won't have pretty icon as previous package did. No functional regression is expected here.

### Option 3 - Request to Apple
If someone in GENIVI org requests to Apple to have a valid App Distribution code-sign key, this will allow DLT Viewer
to be distributed on Release tabs, and etc.

hannesa2, 2021-01-10:
> The build script in `.ci/travis/build.sh` assumes that Qt will live under `/usr/local/opt/lib` which isn't the default location with the latest **Homebrew** installation.
> 
> We can change the script to point out to new **Homebrew**'s folder such as `/opt/homebrew/Cellar/qt/5.12/....`

There is currently no point to change here anything. As long as Github actions use no M1, there is no need to move to arm64 brew usage on CI.
Additionally QT is currently not ready for arm64 https://bugreports.qt.io/browse/QTBUG-85279

For the other thing: I vote for option **3**, but it's out of my control.
For option **2**: can you make a pull request please ? @Elastino

Elastino, 2021-01-11:
I'm sorry but I decided not to make a pull request for it because that script ``.ci/travis/build.sh`` is intended to be used with CI server, and I assume the artifact of this script isn't available for the official release from the first place.

Also, at the moment, I don't have much information how ``homebrew`` is planning to migrate from ``/usr/local/bin`` to ``/opt/homebrew/bin``, I can't simply modify the build script because it will break existing homebrew environment for non-M1 users.

Better approach here if Option 3 is not available to DLT Viewer community, I would go for new formula for DLT viewer in brew's community.

I will update this ticket when dlt-viewer is widely available to common user who just want to use DLT Viewer on Apple silicon.

closed 2021-01-22 by alexmucde