COVESA/dlt-viewer #154: Question regarding FIBEX metadata xml file

MarwanDawood, 2020-12-04:
I am trying to create a FIBEX metadata xml file that will be used to read the DLT messages.
Referring to AUTOSAR_PRS_DiagnosticLogAndTraceProtocol.pdf, I was able to link the frame with the its PDU and signals,
but still the relevant non-static data is not shown into the DLT viewer tool, only the description below is shown.
Is there some missing configuration that should be added? (please see below the xml snippet)

```
<fx:FRAMES>
    <!-- ================== 1. Log and Trace Message ============ -->
    <fx:FRAME ID="ID_1">
        <ho:SHORT-NAME>ID 1</ho:SHORT-NAME>
        <fx:BYTE-LENGTH>1</fx:BYTE-LENGTH>
        <fx:FRAME-TYPE>OTHER</fx:FRAME-TYPE>
        <fx:PDU-INSTANCES>
            <fx:PDU-INSTANCE ID="P 1 0">
                <fx:PDU-REF ID-REF="PDU 1 0"/>
                <fx:SEQUENCE-NUMBER>0</fx:SEQUENCE-NUMBER>
            </fx:PDU-INSTANCE>
            <fx:PDU-INSTANCE ID="P 1 1">
                <fx:PDU-REF ID-REF="PDU 1 1"/>
                <fx:SEQUENCE-NUMBER>1</fx:SEQUENCE-NUMBER>
            </fx:PDU-INSTANCE>
</fx:FRAMES>
<!--*************************** PDUS ***************************-->
<fx:PDUS>
    <!--=============== Message 0   ==================-->
    <!--=============== 1. Parameter ==================-->
    <fx:PDU ID="PDU 1 0">
        <ho:SHORT-NAME></ho:SHORT-NAME>
        <ho:DESC>Temperature: </ho:DESC>
        <fx:BYTE-LENGTH>0</fx:BYTE-LENGTH>
        <fx:PDU-TYPE>OTHER</fx:PDU-TYPE>
    </fx:PDU>
    <!--=============== 1. Parameter ==================-->
    <fx:PDU ID="PDU 1 1">
        <ho:SHORT-NAME></ho:SHORT-NAME>
        <ho:DESC>measurement: </ho:DESC>
        <fx:BYTE-LENGTH>1</fx:BYTE-LENGTH>
        <fx:PDU-TYPE>OTHER</fx:PDU-TYPE>
        <fx:SIGNAL-INSTANCES>
            <fx:SIGNAL-INSTANCE ID="S 1 0">
                <fx:SEQUENCE-NUMBER>0</fx:SEQUENCE-NUMBER>
                <fx:SIGNAL-REF ID-REF="S UINT8"/>
            </fx:SIGNAL-INSTANCE>
        </fx:SIGNAL-INSTANCES>
    </fx:PDU>
</fx:PDUS>
```

alexmucde, 2020-12-11:
Please check the DLT Parser tool https://github.com/GENIVI/dlt-viewer/tree/master/parser,
which is part of the DLT Viewer source code. You can create XML Files for testing.
There are also some examples.

alexmucde, 2021-01-20:
@MarwanDawood Is your problem solved already?

alexmucde, 2021-01-26:
Please reopen or create new ticket if you are not satisfied yet.

closed 2021-01-26 by alexmucde