COVESA/dlt-viewer #167: Improve search speed.

SimonPreisinger, 2021-01-14:
Currently the search is slow with use of a lot of traces and could be improved.
My suggestion is to improve the search speed by using several threads instead of one. 
Should be possible with [Qt Concurrent](https://doc.qt.io/qt-5/qtconcurrent-module.html)
I know there are plugins but they are complicated to install.

Later this can be done for applying a filter also.

alexmucde, 2021-01-20:
We already made some multithreading tests in the past with, but we were not very succesful. I think the bottleneck is the file access to HDD/SSD. But perhaps someone can give it a try again.

SimonPreisinger, 2021-01-23:
I want to try it pls give me access to create a pull request

alexmucde, 2021-01-26:
@SimonPreisinger You can create a clone in your own Github Account, create the change there and then request a Pull Request. You do not need Write access to the DLT Viewer repository to create branches.

svlad-90, 2021-03-15:
Hi @SimonPreisinger, you can avoid inventing a bicycle - try out the DLT-Message-Analyzer plugin: https://github.com/svlad-90/DLT-Message-Analyzer. It covers your need as it creates the in-RAM-cache of all the messages within the file ( or the ones which are coming in from the opened connection to a target device ) and searches within the cache. The size of the cache is adjustable. The search is multi-threaded with the possibility to select the number of the threads to be used. The speed of the search is in average ~10 times faster.

P.S. I'm not trying to convince you to not extend the core functionality. Just telling, that the way exist, which might save you some efforts ))

alexmucde, 2021-03-31:
Any progress on this topic?

alexmucde, 2021-11-23:
As there is no progress i will close this issue. New issue exists to use a search database engine for faster search in milliseconds. See  #202