COVESA/dlt-viewer #172: Downloadable artifacts in Release

hannesa2, 2021-01-22:
When you tell me what the artifacts for each/some platform/s I can make you an action on Github which provides it here https://github.com/GENIVI/dlt-viewer/releases

At least for macOS it's not a single file (and signing is an issue)

Here is an example https://github.com/ytai/ioio/releases/tag/6.1.0

lukx, 2021-01-22:
I don't know whether this answers (part of) your question but:
"DLT SDK" is the dlt-viewer binary package for windows, 64 bit
"DLT Parser" is the dlt-parser binary package for windows, 64 bit

hannesa2, 2021-01-22:
When  you go to pull request job detail (not release) eg https://github.com/GENIVI/dlt-viewer/runs/1747384132
you have the snapshots already there 

[image]

I can only test macOS, and there I've problems 
[image]

I can only confirm for macOS it's not working now 👎 

@lukx 
Please can you test on Windows the artifacts ?

Who can do this for Linux ?

alexmucde, 2021-01-22:
Can we also use Github Actions to build on master the latest commit, as Appveyor and Travis did? I am currently downloading the Artifact, download is quiet slow.

lukx, 2021-01-22:
@hannesa2 downloaded dlt-sdk from the run you linked, it starts normally (but I have no log available to actually test the programm)

alexmucde, 2021-01-22:
Yes it works, also with log file.

[image]

hannesa2, 2021-01-22:
> Can we also use Github Actions to build on master the latest commit, as Appveyor and Travis did?

Here you go #176

alexmucde, 2021-01-22:
Appveyor still builds, perhaps i have to delete the two remaining Webhooks in the project settings?

[image]

alexmucde, 2021-01-22:
Deleted some old Appveyor jobs in my Appveyor account, should not build anymore.

alexmucde, 2021-01-26:
Are we finished with this issue?

hannesa2, 2021-01-26:
You mean the "Download-able artifacts in Release" ?
No, I've no feedback from Linux and macOS doesn't work

But I could do it in the meantime for Windows only

hannesa2, 2021-02-18:
### Linux
I checked on a Linux machine and generated Linux artifact starts. But you have to gave it `+x` permission in advance.
Maybe I will make a pull request with this last step.

### Release with artifacts

As long as we build release at least on 3 different os, I run into the issue that I've to collect from 3 machine the artifacts, but currently I don't now how to do that.
Collection in a pull request job is quite easy and already implemented.

hannesa2, 2021-03-03:
I close this, because there is a pull request #184

closed 2021-03-03 by hannesa2