COVESA/dlt-viewer #177: No log displayed after clicking Clear log

OXINARF, 2021-01-22:
I regularly hit this use-case:
- I'm connected to an ECU (via TCP, in case it matters)
- I reboot the ECU
- While it is rebooting, I clear the temporary log (and create a new one) by clicking the Clear log button on the toolbar
- When the ECU boots it starts sending the log (as I can see the version and received packets increasing), but there is no message displayed
- The way to recover is making a change to configuration and applying it

I've found this: https://github.com/GENIVI/dlt-viewer/blob/86f21387204783fd57ba115528f440314c8fd4bf/src/mainwindow.cpp#L1818-L1819 But it doesn't seem enough for the log to not display at all.

alexmucde, 2021-01-26:
Which version of DLT Viewer are you using? Can you try lattest version on github?

OXINARF, 2021-02-03:
Hi Alex,

I should have written that, it would clarify it to me too 😄

I opened this issue after I installed the latest 2.21.1 because I thought I tested it and the problem was still there. However, after many tries, it is working fine - so now I think I still had the 2.20.2 version opened and I mistakenly tested there instead of the latest version.

Sorry for the disturbance 🙂

closed 2021-02-03 by OXINARF