COVESA/dlt-viewer #178: Publish DLT-Viewer on Flathub

AlessandroBono, 2021-02-08:
### What is Flathub?
Flathub aims to be the place to get and distribute apps for Linux. It is powered by Flatpak which allows Flathub apps to run on almost any Linux distribution (see [1]).

### What is Flatpak?
Flatpak is a next-generation technology for building and distributing desktop application on Linux (see [2]).

### Benefits of Flatpak/Flathub
This is a very broad topic. I'll try to summarize the benefits from a developer point-of-view but other benefits exist. The applications distributed as a Flatpak are consistent with the toolchain of the developer rather than the distribution. That means that the user gets the same application that the developer has built which makes it a lot easier to figure out bugs. Distributions at times change code of the application when porting to their toolchain - this may result in different bugs on different distributions. With Flatpak the developer is in control of all the stack by choosing the runtime and bundling missing libraries if needed. Furthermore, by using Flathub, the developer is in control of releasing new versions: when a new version is released everyone will get the new release independently of the distribution used or how old the distribution is. For a list of supported distribution see [3].

### Current situation
I wrote a Flatpak manifest for DLT-Viewer that you can find [here](https://github.com/AlessandroBono/dlt-viewer-flatpak). It works. The major downside is that, due to the sandbox, the serial connections are not working out-of-the-box yet (see https://github.com/AlessandroBono/dlt-viewer-flatpak/issues/1). What is missing is a serial port portal that let the application ask to the OS access for the serial device under the user consent. If you want to test it, just follow the instructions in the `README.md` file. Keep in mind that once published on Flathub such steps are not required and the user don't have to install `org.kde.Sdk`. Only `org.kde.Platform` will be downloaded automatically when installing DLT-Viewer if not already present on the system (runtimes are shared across applications). 

### Why this issue?
Flathub is primary meant for upstream developers. [It is advised](https://github.com/flathub/flathub/wiki/App-Submission) to contact the upstream project for submitting applications on Flathub. A submission is ready [here](https://github.com/AlessandroBono/flathub) but I won't go forward if there is no interest from this project to take ownership of Flathub releases. [As you can see](https://github.com/AlessandroBono/dlt-viewer-flatpak/commit/4b7f3a1e2c7a175a5297f9ec267bc497fa59b5a8) making a new release is just about changing the download URL, the SHA256 of the archive and update the metainfo data by adding a new release tag. I could co-maintain the manifest and give my help on understanding how Flatpak works, but the final decision on publishing and maintaining releases on Flathub is up to you.

Let me know what you think!

[1] https://flathub.org/about
[2] https://flatpak.org
[3] https://flatpak.org/setup

hannesa2, 2021-03-03:
You know what to do, what's about a pull request ? 
I guess the maintainer is happy to approve any pull request.

alexmucde, 2021-03-03:
Hello @AlessandroBono . Yes please do the next steps which are needed to bring DLT Viewer to flathub. Should we do something or will you proivde a pull request?

AlessandroBono, 2021-03-19:
DLT Viewer is now on Flathub. You can install it on any [Flatpak/Flathub configured systems](https://flatpak.org/setup/) via GUI application such as GNOME Software or by executing the following command:
```
flatpak install flathub org.genivi.DLTViewer
```
@alexmucde did you receive a collaboration invite for https://github.com/flathub/org.genivi.DLTViewer?

closed 2021-03-19 by AlessandroBono