COVESA/dlt-viewer #179: Mark/Unmark lines function doesn't work on log messages highlighted by a filter

4og, 2021-02-10:
**Steps to reproduce**
1. Add a filter with a marker color.
2. Select any message highlighted by the filter.
3. Choose "Mark/Unmark line(s)" in the pop up menu appeared by the right click.
4. Observe that nothing happens.

**Expectation**

The "Mark/Unmark line(s)" overrides the color from the filter.

Used DLT Viewer version: 86f21387204783fd57ba115528f440314c8fd4bf

alexmucde, 2021-03-03:
@4og I repoduced the issue. Looks like the priority of marking logs is wrong. Manual marking should have a higher priority.
Are you able to provide a pull request to fix this?

alexmucde, 2021-03-31:
Provided Pull Request. Marks have now the highest priority.

closed 2021-03-31 by alexmucde