COVESA/dlt-viewer #196: ../../qdlt/qdltfilter.h:34:10: fatal error: QRegularExpression: No such file or directory

serkazi, 2021-03-31:
I followed the steps in `INSTALL.md` for my Ubuntu 18.04 box. However, after the `qmake` step, the `make` produces the error in the subject. Did I do something wrong? All the listed dependencies are installed.

alexmucde, 2021-03-31:
@serkazi Are you sure, that you have installed the right Qt5 Headers? The class was introduced with Qt5, Qt4 will not work.

serkazi, 2021-03-31:
Somehow, the `qmake` step that is mentioned in `INSTALL.md` happens to erroneously capture the Qt4 (that is installed alongside Q5 on my system). Long story short, I then noticed that there is a `CMakeLists.txt` and I used the standard `cmake .. && make` procedure. Now it works. Thanks!