COVESA/dlt-viewer #197: DLT-Viewer: How to apply parser to decode messages (Non Verbose Mode Plugin)

GirijaG, 2021-04-07:
Hi Team,

I need to do below actions in command line
[image]

To do so, I used below command at command prompt (Windows 10).
dlt_viewer.exe DLT_18_2_20.dlt -e "Non Verbose Mode Plugin|D:/DltViewer/DLT_PARSER_OUTPUT.xml|Apply Configuration"

But getting below error: 
[image]

Please help me out which command to use, to convert .dlt files to .csv/.txt files by applying the parser.

Without applying parser, below commands works:

1. dlt_viewer.exe -csv -c DLT_0_19_72_2.dlt dlt_csv.csv
2. dlt_viewer.exe -u -c DLT_0_19_72_2.dlt dlt_text.txt

Thanks and Regards
Girija

alexmucde, 2021-04-12:
@GirijaG You cannot use the -e option to provide configuration files to plugins.
You have to load and activate a plugin in the DLT Viewer and store the configuration into a project file.
Then you can load the project by command line with the -p option.

Does this solve the issue?

GirijaG, 2021-04-12:
> @GirijaG You cannot use the -e option to provide configuration files to plugins.
> You have to load and activate a plugin in the DLT Viewer and store the configuration into a project file.
> Then you can load the project by command line with the -p option.
> 
> Does this solve the issue?

Yes, thank you, 

dlt-viewer.exe -p dlt-parser.dlp -s -u -c DLT_0_19_73_3.dlt dlt.txt

the above command i m using to convert and it is working.

closed 2021-04-12 by GirijaG

Znzero, 2023-02-15:
> 

how to  load and activate a plugin in the DLT Viewer and store the configuration into a project file?