COVESA/dlt-viewer #202: Use of a search engine library to improve speed

alexmucde, 2021-04-23:
A search engline library should be used to improve search speed and filtering in the DLT Viewer.

One possible library is Lucene++, where already a prototype exists.

Further investigations needed like this could be integrated best like in a Plugin or in the core.

sherlok235, 2024-10-03:
I would like to help with this task, it seems interesting.

alexmucde, 2024-10-11:
@sherlok235 I have assigned you the issue. Would be good to have this optional, e.g. in a plugin.

alexmucde, 2024-10-11:
It also important that the dlt-viewer can be still build without a depent library like lucene++

alexmucde, 2024-10-11:
In Ubuntu 24.04 i have found a package for lucence liblucene++-dev

sherlok235, 2024-10-11:
> @sherlok235 I have assigned you the issue. Would be good to have this optional, e.g. in a plugin.

Ok,  I'll study how I can do it.

sherlok235, 2024-10-11:
> It also important that the dlt-viewer can be still build by defualt without a depent library like lucene++

I think I can make that work if I add an option to the qmake file.

sherlok235, 2024-10-26:
@alexmucde I need some test data for testing, I don't have it yet and would like to ask you to provide it. And I am also interested if there is an application that can receive the test data and emulate the data transfer between the ECU and the DLT viewer (to emulate delays).