COVESA/dlt-viewer #205: [Windows] High DPI Displays and scale factor != 100%

gobater, 2021-04-27:
In the last releases Qt has been introducing support for high DPI Displays (https://doc.qt.io/qt-5/highdpi.html). According to the documentation, High DPI support is enabled from Qt 5.6 onward

In a Windows PC, in high DPI displays when using scaling, DLT-viewer looks weird. The Qt Widgets seem to be scaled twice: once by the system, and another by Qt.
The screenshot below compares DLT-Viewer 2.16 (Qt 5.5) with DLT-Viewer 2.21 (Qt 5.14) and windows explorer in a high DPI display (4k resolution, 27") with 150% scaling (no photoshop: the three windows are rendered in the same screen at the same time).

Compare the text size and icons in the three windows. Note also some rendering artifacts in old DLT-Viewer
[image]

The problems become worse in combination with multiple displays and different scale factors.

This might seem a Qt issue, and not related to DLT-Viewer. However, making the application High DPI and scaling aware could improve rendering (similar experience to Ms. Windows) in all kind of combinations (multiple monitors and different scale factors).

According to the Qt documentation about HighDpiScaleFactorRoundingPolicy (https://doc.qt.io/qt-5/qt.html#HighDpiScaleFactorRoundingPolicy-enum) and the results observed rounding seems to apply for non integer scaling factors (like 150%) making Qt to scale (again) the widgets. Setting the environment variable `QT_SCALE_FACTOR_ROUNDING_POLICY` to "Passthrough" before starting Dlt-Viewer makes the DLT-Viewer to look nice again in all monitors (also with different scaling).

Comparison of Dlt-Viewer in a high DPI display (4k resolution, 27") with 150% scaling and Passthrough rendering policy.
[image]

alexmucde, 2022-05-07:
I see same issues also when switching between different monitors with different resolutions. Some Windows applications automatically detects this and changes the resolution. Also a zoom feature would be nice, especially useful during presentations.
Any contributers here to work on this?

alexmucde, 2024-02-23:
@gobater Still an issue? Very old ticket.

gobater, 2024-02-28:
@alexmucde , yes still present on 2.25.0 looks awful on my 150% scaled monitor.

Setting the environment variable *QT_SCALE_FACTOR_ROUNDING_POLICY* to the right value (Passthrough) makes dlt-viewer to look how it shall.

alexmucde, 2025-07-24:
@gobater Still an issue in latest Qt6 versions?

Elastino2, 2026-10-27:
# DLT : 2.31.1 with Qt6.12 on Windows11

## Scale 150%
[image]

## Scale 200%
[image]

@alexmucde It seems like okay for now.