COVESA/dlt-viewer #217: Feature to filter messages by/between some index

SvenLuebke, 2021-07-21:
Dear dlt-viewer developers!

Is it somehow possible to filter messages by index? First I thought, it's the filter option "Message Id" but that is something different. Of course I can cut & export the file, but that would be an additional manual step.

Background: I have a Dlt log which captured two lifecycles of a Dlt device and within this file a reboot happened. I'd like to view (and sort timestamps for) messages after the reboot and not before.

Thank you for your information!

BR,
Sven

alexmucde, 2021-07-22:
@SvenLuebke Interesting feature, you think it should be a simple filter, where you can select first and last message or it is more like a negativ filter, where you do not want see the messages before the first selected and after the last.

SvenLuebke, 2021-07-23:
@alexmucde A simple filter would be enough, if other filters can process the output of that message index filter and don't process the original Dlt file. I guess it's currently not working like that, right?

For me it would be important, that the setting for the index range is at one point only, so I don't have to adapt the this range for 10 filters.

If that feature won't be assigned to a filter, it would be a good idea to put the index range right under the "Sorted by Time" "Sorted by Timestamp" checkboxes (of Project tab) and make it dis-/enableable.

alexmucde, 2021-11-23:
It should be perhaps not only a filter, it should be more a general setting to improve the search operation, Only messages inside the range should be processed. In addition it should be also possible to select a time range. Title of issue updated.

alexmucde, 2022-05-07:
Any contributor available for this very interesting feature tô provide a PR? This would really speed up the daily work, when working with big dlt files.

SvenLuebke, 2022-11-24:
Thank you very much @Manik-ltts and @alexmucde for implementing this feature!