COVESA/dlt-viewer #218: QT5 build on MacOS

rafaelpirolla, 2021-07-28:
Just as documentation:

cmake .. -DCMAKE_PREFIX_PATH=/usr/local/Cellar/qt@5/5. whatever version brew install qt@5 is nowadays.

rafaelpirolla, 2021-07-29:
I'm unable to run it with any plugins though as the application just crashes on start. Saw that someone had a patch in the mailing list in the past but it's not there anymore.

chrsch, 2021-08-03:
On Mac just checkout and do

- brew install qt@5
- mkdir build && cd build
- /usr/local/opt/qt@5/bin/qmake ../BuildDltViewer.pro
- make

You'll find the binary in build/release

alexmucde, 2021-08-16:
@rafaelpirolla Is this issue solved or still open? Sorry i cannot help, I am not familiar with MacOS build.

Elastino, 2021-08-22:
I believe OP uses Apple M1 silicon, and is trying to use CMAKE to build DLT-Viewer.
Building DLT-Viewer with CMake, used be working fine which results DLT-Viewer.App via Ninja package.
Somehow, CMake is broken on Ninja build where failing to link valid Qt5 libs. 
Please use QMake approach described above. 
This is out of scope for DLT project for supporting Cmake on Apple M1. (Although I work on Apple M1. :( So sad.)

alexmucde, 2021-11-23:
Look like issue solved, reopen if not.