COVESA/dlt-viewer #220: Serial connection disconnects/reconnects when log file is rotated.

Bengineer2048, 2021-08-13:
When specifying a Maximum file size in settings and logging VuC through a serial connections the connection briefly dropped when the log file is rotated.

alexmucde, 2021-08-16:
@Bengineer2048 I have checked the source code and it looks like the old file is copied first and then a new file is created. Normally this should have no influence on your serial connection. In a TCP connection a timeout can be configured, but not for serial connections. So even a slow PC should have no influence, even copying a file is not really a good implementation.

Please provide more information like which connections you are using in parallel.

These are the functions which are called when splitting the file:

`
void MainWindow::createsplitfile()
{
    // get new filename
    dltIndexer->stop();
    QFileInfo info(outputfile.fileName());

    QString newFilename = info.baseName()+
            (startLoggingDateTime.toString("__yyyyMMdd_hhmmss"))+
            (QDateTime::currentDateTime().toString("__yyyyMMdd_hhmmss"))+
            QString(".dlt");
    QFileInfo infoNew(info.absolutePath(),newFilename);
    qDebug() << "Split to" <<  outputfile.fileName() << "to" << infoNew.absoluteFilePath();

    // rename old file
    outputfile.copy(outputfile.fileName(),infoNew.absoluteFilePath());

    // set new start time
    startLoggingDateTime = QDateTime::currentDateTime();

    SplitTriggered(info.absoluteFilePath());

}

void MainWindow::SplitTriggered(QString fileName)
{
    // change DLT file working directory
    workingDirectory.setDltDirectory(QFileInfo(fileName).absolutePath());

    // close existing file
    if(outputfile.isOpen())
    {
        //qDebug() << "isOpen" << fileName << __FILE__ << __LINE__;
        if (outputfile.size() == 0)
        {
            deleteactualFile();
        }
        else
        {
            outputfile.close();
        }
    }

    // create new file; truncate if already exist
    outputfile.setFileName(fileName);
    setCurrentFile(fileName);

    outputfileIsTemporary = false;
    outputfileIsFromCLI = false;

    if(true == outputfile.open(QIODevice::WriteOnly|QIODevice::Truncate))
     {
        openFileNames = QStringList(fileName);
        isDltFileReadOnly = false;
        reloadLogFile(false,true);
     }
    else
     {
        if (QDltOptManager::getInstance()->issilentMode())
         {
         qDebug() <<  QString("Cannot create new log file ") << outputfile.fileName() << fileName << outputfile.errorString();
         }
         else
         {
         QMessageBox::critical(0, QString("DLT Viewer"),
                                  QString("Cannot create new log file \"%1\"\n%2")
                                  .arg(fileName)
                                  .arg(outputfile.errorString()));
         }
     }
 }
`

Bengineer2048, 2021-08-19:
@alexmucde Thank you so much for reviewing my request and reaching out! 
I know more about the issue now and can provide more detail. 
For my application I am logging for TCP and serial connections, however, I tested this with just a serial connection and observed the same issue when the file is split after the size limit is reached.

I found in the code where this is happening, but I'm sure if it's intentional. 

In the function reloadLogFile()  there is:
    // prevent further receiving any new messages
    saveAndDisconnectCurrentlyConnectedSerialECUs(); 

saveAndDisconnectCurrentlyConnectedSerialECUs is what is calling the disconnect for the serial interface during the split. It's not clear to me why this function needs to be called. Specially it's odd that serial connections are being disconnected but not TCP connections. My first thought would be to just comment out this line but I don't want to do that without understanding the purpose.

I also found that at the beginning of the reloadLogFile function there is:
if( 0 != settings->loggingOnlyMode )
    {
        qDebug() << "Logging only mode !";
       // return;
    }

For my use case I am using logging mode so I just uncommented the return here to avoid ever getting to the saveAndDisconnectCurrentlyConnectedSerialECUs() call in the first place and it worked. But it would be great to find a solution that works while not in logging mode.

alexmucde, 2021-08-20:
@Bengineer2048 Thanks for the deeper analyses. I cannot remember why the call saveAndDisconnectCurrentlyConnectedSerialECUs(); was introduced. It makes also no sense to me. It is even worse, as we will loose the buffered data. Perhaps there were some issues/crashes in the past, but this should not be solved in this way.

Can you try to remove the following calls: 

saveAndDisconnectCurrentlyConnectedSerialECUs();.
connectPreviouslyConnectedECUs();

and test if you have any issues. Also Reloading the DLT File, during an open connection should be tested, e.g. when you change the filter.

Bengineer2048, 2021-08-23:
@alexmucde I did as you suggested and removed those two calls. I changed the filters several times during logging to reload the DLT file and everything worked as expected. I did not notice any issues related to removing those calls. Note: I only tested on Linux.

closed 2021-11-23 by alexmucde